mohamed-badaoui/asus-touchpad-numpad-driver #1: Asus zenbook ux434FA-linux mint ulyssa os 5.10.0-051000-generic

cedrichappy, 2021-04-02:
hello,
when i run the command i've:

i2c-3	 i2c          i915 gmbus dpb                  	                I2C adapter
i2c-1	 i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-6	 i2c       	i915 gmbus dpd                  	                I2C adapter
i2c-4	 i2c       	i915 gmbus dpc                  	                I2C adapter
i2c-2	 i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-0	 smbus    SMBus I801 adapter at efa0                   SMBus adapter
i2c-7	 i2c       	AUX A/port A                    	               I2C adapter
i2c-5	 i2c       	i915 gmbus misc                 	               I2C adapter

I would like to know what to choose and where exactly to change the name in the .py?

And then the command that could work with the new name and number?

I just bought this laptop and I removed windows but everything works except the num pad (touch pad work)

Thank you for your work and your time.

mohamed-badaoui, 2021-04-02:
Hi,
I would say that you need no code modifications. That model use a pretty similar  to the ux433 model.

So ensure that you have python3 well configured with required packages and just run the install.sh

cedrichappy, 2021-04-02:
ok thaks i've installed packages and i run the command sudo i2ctransfer.......and the num pad turn on and install .sh and reboot but 
the calculator shortcut does not work and fn + f8 does not work to turn on the num pad

cedrichappy, 2021-04-02:
sorry my fn+ech key was activated so I deactivated it and I press fn f8 and it lights up but when I try it on a text editor for example it doesn't display a number

mohamed-badaoui, 2021-04-03:
Ok nice. Some updates were made.
Can you download las version of python program (ux433_touchpad_numpad.py) and run the install again (sh install.sh)

You do not need to reboot and try again.

cedrichappy, 2021-04-03:
hello,
i'm update python program  and run sudo sh install.sh(Result: i2c-dev in my terminal) 
But no works

mohamed-badaoui, 2021-04-03:
So that means your laptop model works slightly differently.
I invite you to follow the discussion thread below and ask your question to the community.
Maybe someone has faced your problem already and found a solution.
What I know is that it works for the following 2 models:
- Asus UX433
- Asus M433IA

https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1810183

Hope you find a solution

mohamed-badaoui, 2021-04-03:
Hi Cedric,

Can you send me the results of that commands to know the elan devices you have on your machine ?

```
grep -B1 -A5  "ELAN" /proc/bus/input/devices
```
probably your touchpad is not correctly found.

cedrichappy, 2021-04-03:
hi,

uname -a    5.10.0-051000-generic 

grep -B1 -A5  "ELAN" /proc/bus/input/devices:

I: Bus=0018 Vendor=04f3 Product=2a5c Version=0100
N: Name="ELAN9008:00 04F3:2A5C"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ELAN9008:00/0018:04F3:2A5C.0001/input/input19
U: Uniq=
H: Handlers=mouse0 event6 
B: PROP=2
B: EV=1b
B: KEY=400 0 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=2a5c Version=0100
N: Name="ELAN9008:00 04F3:2A5C UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ELAN9008:00/0018:04F3:2A5C.0001/input/input20
U: Uniq=
H: Handlers=event7 
B: PROP=0
B: EV=9
B: ABS=10000000000
--
I: Bus=0018 Vendor=04f3 Product=2a5c Version=0100
N: Name="ELAN9008:00 04F3:2A5C UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ELAN9008:00/0018:04F3:2A5C.0001/input/input21
U: Uniq=
H: Handlers=event16 
B: PROP=0
B: EV=100001

this is a picture taken on the internet to show you what my num pad looks like

[image]

mohamed-badaoui, 2021-04-03:
I see that the key layout is not the same as mine (UX433) but anoter contributor (magellan-2000) on launchpad has the same key layout.
Feel free to contact him directly.
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1810183

cedrichappy, 2021-04-03:
ok thanks !

mohamed-badaoui, 2021-04-04:
Hi cedric,

Your issue is that your touchpad is not correctly detected.
Perhaps your touchpad is not referenced as ELAN.

If you print the result of that command
grep -B1 -A5 -i -E "(ELAN|asus)" /proc/bus/input/devices 
 
See if there is a touchpad devices named asus instead Elan.

cedrichappy, 2021-04-04:
hi mohamed,

here is the return of the order:

I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input11
U: Uniq=
H: Handlers=rfkill kbd event4 
B: PROP=0
B: EV=100013
B: KEY=1000000080000 0 0 0 0 181606f00900000 8280027801701000 e000000000000 0
--
I: Bus=0018 Vendor=04f3 Product=2a5c Version=0100
N: Name="ELAN9008:00 04F3:2A5C"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ELAN9008:00/0018:04F3:2A5C.0001/input/input12
U: Uniq=
H: Handlers=mouse0 event5 
B: PROP=2
B: EV=1b
B: KEY=400 0 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=2a5c Version=0100
N: Name="ELAN9008:00 04F3:2A5C UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ELAN9008:00/0018:04F3:2A5C.0001/input/input13
U: Uniq=
H: Handlers=event6 
B: PROP=0
B: EV=9
B: ABS=10000000000
--
I: Bus=0018 Vendor=04f3 Product=2a5c Version=0100
N: Name="ELAN9008:00 04F3:2A5C UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ELAN9008:00/0018:04F3:2A5C.0001/input/input14
U: Uniq=
H: Handlers=event10 
B: PROP=0
B: EV=100001

--
I: Bus=0018 Vendor=04f3 Product=310d Version=0100
N: Name="ASUE1407:00 04F3:310D Mouse"
P: Phys=i2c-ASUE1407:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE1407:00/0018:04F3:310D.0002/input/input17
U: Uniq=
H: Handlers=mouse1 event8 
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=310d Version=0100
N: Name="ASUE1407:00 04F3:310D Touchpad"
P: Phys=i2c-ASUE1407:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE1407:00/0018:04F3:310D.0002/input/input18
U: Uniq=
H: Handlers=mouse2 event9 
B: PROP=5
B: EV=1b
B: KEY=e520 10000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=310d Version=0100
N: Name="ASUE1407:00 04F3:310D Keyboard"
P: Phys=i2c-ASUE1407:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE1407:00/0018:04F3:310D.0002/input/input19
U: Uniq=
H: Handlers=sysrq kbd event11 leds 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff800000000007ff febeffdfffefffff fffffffffffffffe

mohamed-badaoui, 2021-04-04:
Ok nice, I see your issue.
As I said, it is because of a bad touchpad detection.

I update the version of your driver. Re download the project and make the installation with
sudo ./install.sh

then choice 2

It should works now, If so, do not hesitate to let a comment to the launchpad community to identify all asus version compatible with that python driver.

cedrichappy, 2021-04-04:
I just deleted the old .py

mohamed-badaoui, 2021-04-04:
all depends on how you did the previous installation.
If you were just testing the py script without having launched the install.sh, so you have no more things to do.
Else if install.sh was executed  so a cleanup is necessary.

cedrichappy, 2021-04-04:
sudo: ./install.sh : command not found  ?

maybe sudo sh install.sh ?

mohamed-badaoui, 2021-04-04:
no it is 
sudo ./install.sh

Are you in the right folder?

cedrichappy, 2021-04-04:
ok nice!!!!!

chmod + x install.sh and it's work !

cedrichappy, 2021-04-04:
many many  thanks mohamed!!!!!

mohamed-badaoui, 2021-04-04:
You re welcome Cedric !
Enjoy your new Numpad ^ ^

Please let a comment in launchpad discussion.
Thanks
I close that issue.

closed 2021-04-04 by mohamed-badaoui

cedrichappy, 2021-04-04:
ok i will make a comment and i have a last question is the counter of the cursor working at the same time as the numeric keypad because for me no i can't move my mouse when the numeric keypad is activated

mohamed-badaoui, 2021-04-04:
You cannot use both numpad and touchpad at the same time

cedrichappy, 2021-04-04:
ok many thanks!!

cedrichappy, 2021-04-07:
hello mohamed,

I reinstalled my system so I'm on linux mint xfce os 5.8 when I reinstall driver it doesn't work and when I try to uninstall it as you show me it gives me this

Testing interface i2c-1 : failed
Testing interface i2c-7 : sucess
Does your numpad has % and = symbols [N/y]y
Your numpad has % and = symbols.

What is your keyboard layout?
1) Qwerty
2) Azerty
3) Quit
Please enter your choice [1-3]: 2
Copy asus python driver to /usr/bin/asus_touchpad_numpad.py
Add asus touchpad service in /lib/systemd/system/
Asus touchpad service enabled
Asus touchpad service started

and when i remove

Failed to stop ux433_touchpad_numpad.service: Unit ux433_touchpad_numpad.service not loaded.

Failed to disable unit: Unit file ux433_touchpad_numpad.service does not exist.

mohamed-badaoui, 2021-04-07:
Hello Cedric,

Are you sure to have the last version? because there is no more **ux433_touchpad_numpad.service**
The service is named **asus_touchpad_numpad.service**

Check this first.

cedrichappy, 2021-04-07:
hi,
i'm install and:

when I place this order this is why it doesn't work:  sudo systemctl status asus_touchpad_numpad.service

● asus_touchpad_numpad.service - Asus Touchpad to Numpad Handler
     Loaded: loaded (/lib/systemd/system/asus_touchpad_numpad.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2021-04-07 17:47:39 CEST; 5min ago
    Process: 11019 ExecStart=/usr/bin/python3 /usr/bin/asus_touchpad_numpad.py (code=exited, status=1/FAILURE)
   Main PID: 11019 (code=exited, status=1/FAILURE)

avril 07 17:47:38 ZenBook systemd[1]: Started Asus Touchpad to Numpad Handler.
avril 07 17:47:39 ZenBook systemd[1]: asus_touchpad_numpad.service: Main process exited, code=exited, status=1/FAILURE
avril 07 17:47:39 ZenBook systemd[1]: asus_touchpad_numpad.service: Failed with result 'exit-code'.
~
~
~

cedrichappy, 2021-04-07:
Maybe xfce setting?
[image]

mohamed-badaoui, 2021-04-07:
Hi Cedric,

The touchpad detection for keyboard layout having symbols has been fixed.
Please download last version and re run the install.sh

It should work now.

cedrichappy, 2021-04-07:
Yeah!!!!! it s works many thanks Mohamed for your time!!