mohamed-badaoui/asus-touchpad-numpad-driver #8: Support for Asus Zenbook Flip 13 UX363EA-IH74T

hjheins, 2021-04-10:
Hi,
I have the laptop mentioned in the title. It has the exact same touchpad layout and seems to support a numpad within the touchpad (in Windows).
Would it be possible to add support for this laptop in your code?
I tested with the existing code on Arch, but I can not get this to work reliably in its current form.
Attached the output showing the devices.
Thanks.

Hendrik-Jan

hjheins, 2021-04-10:
Attach did not work. The command+output:

# grep -B1 -A5 -i -E "(elan|asu)" /proc/bus/input/devices
I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input6
U: Uniq=
H: Handlers=kbd event6 rfkill 
B: PROP=0
B: EV=100013
B: KEY=1000000080000 0 0 0 0 181606f00900000 8280027801701000 e000000000000 0
--
I: Bus=0018 Vendor=04f3 Product=2d4b Version=0100
N: Name="ELAN9008:00 04F3:2D4B"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-10/i2c-ELAN9008:00/0018:04F3:2D4B.0002/input/input20
U: Uniq=
H: Handlers=event8 mouse0 
B: PROP=2
B: EV=1b
B: KEY=400 0 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=2d4b Version=0100
N: Name="ELAN9008:00 04F3:2D4B UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-10/i2c-ELAN9008:00/0018:04F3:2D4B.0002/input/input21
U: Uniq=
H: Handlers=event9 
B: PROP=0
B: EV=9
B: ABS=10000000000
--
I: Bus=0018 Vendor=04f3 Product=2d4b Version=0100
N: Name="ELAN9008:00 04F3:2D4B UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-10/i2c-ELAN9008:00/0018:04F3:2D4B.0002/input/input22
U: Uniq=
H: Handlers=event10 
B: PROP=0
B: EV=100001

I: Bus=0018 Vendor=04f3 Product=2d4b Version=0100
N: Name="ELAN9008:00 04F3:2D4B"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-10/i2c-ELAN9008:00/0018:04F3:2D4B.0002/input/input23
U: Uniq=
H: Handlers=event11 mouse1 
B: PROP=2
B: EV=1b
B: KEY=c03 0 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=2d4b Version=0100
N: Name="ELAN9008:00 04F3:2D4B UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-10/i2c-ELAN9008:00/0018:04F3:2D4B.0002/input/input25
U: Uniq=
H: Handlers=event12 
B: PROP=0
B: EV=9
B: ABS=10000000000
--
I: Bus=0018 Vendor=04f3 Product=2d4b Version=0100
N: Name="ELAN9008:00 04F3:2D4B UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-10/i2c-ELAN9008:00/0018:04F3:2D4B.0002/input/input26
U: Uniq=
H: Handlers=event13 
B: PROP=0
B: EV=9
B: ABS=10000000000
--
I: Bus=0018 Vendor=04f3 Product=2d4b Version=0100
N: Name="ELAN9008:00 04F3:2D4B UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-10/i2c-ELAN9008:00/0018:04F3:2D4B.0002/input/input27
U: Uniq=
H: Handlers=event17 
B: PROP=0
B: EV=9
B: ABS=10000000000
--
I: Bus=0018 Vendor=04f3 Product=3134 Version=0100
N: Name="ASUE140A:00 04F3:3134 Mouse"
P: Phys=i2c-ASUE140A:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-11/i2c-ASUE140A:00/0018:04F3:3134.0003/input/input30
U: Uniq=
H: Handlers=event15 mouse2 
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=3134 Version=0100
N: Name="ASUE140A:00 04F3:3134 Touchpad"
P: Phys=i2c-ASUE140A:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-11/i2c-ASUE140A:00/0018:04F3:3134.0003/input/input31
U: Uniq=
H: Handlers=event16 mouse3 
B: PROP=5
B: EV=1b
B: KEY=e520 10000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=3134 Version=0100
N: Name="ASUE140A:00 04F3:3134 Keyboard"
P: Phys=i2c-ASUE140A:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-11/i2c-ASUE140A:00/0018:04F3:3134.0003/input/input32
U: Uniq=
H: Handlers=sysrq kbd leds event18 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff800000000007ff febeffdfffefffff fffffffffffffffe
--
I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="Asus Touchpad/Numpad"
P: Phys=
S: Sysfs=/devices/virtual/input/input40
U: Uniq=
H: Handlers=kbd event24 
B: PROP=0

mohamed-badaoui, 2021-04-11:
Hi,
What is the output of this

```
sudo aptitude install libevdev2 i2c-tools
sudo modprobe i2c-dev
sudo i2cdetect -l
sudo dpkg -l| grep libevdev
```

hjheins, 2021-04-11:
Hi,

I think this is the time I should tell you I am running Arch Linux :-)
The outputs:

[code]
# i2cdetect -l
i2c-3   i2c             i915 gmbus tc1                          I2C adapter
i2c-10  i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-1   i2c             i915 gmbus dpb                          I2C adapter
i2c-8   i2c             i915 gmbus tc6                          I2C adapter
i2c-6   i2c             i915 gmbus tc4                          I2C adapter
i2c-13  i2c             AUX USBC1/DDI TC1/PHY TC1               I2C adapter
i2c-4   i2c             i915 gmbus tc2                          I2C adapter
i2c-11  i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-2   i2c             i915 gmbus dpc                          I2C adapter
i2c-0   i2c             i915 gmbus dpa                          I2C adapter
i2c-9   smbus           SMBus I801 adapter at efa0              SMBus adapter
i2c-7   i2c             i915 gmbus tc5                          I2C adapter
i2c-14  i2c             AUX USBC2/DDI TC2/PHY TC2               I2C adapter
i2c-5   i2c             i915 gmbus tc3                          I2C adapter
i2c-12  i2c             AUX A/DDI A/PHY A                       I2C adapter
[/code]

and:
[code]
# pacman -Q -i libevdev
Name            : libevdev
Version         : 1.11.0-1
Description     : Wrapper library for evdev devices
Architecture    : x86_64
URL             : https://www.freedesktop.org/wiki/Software/libevdev/
Licenses        : custom:MIT
Groups          : None
Provides        : libevdev.so=2-64
Depends On      : glibc
Optional Deps   : None
Required By     : libinput  python-libevdev
Optional For    : None
Conflicts With  : None
Replaces        : None
Installed Size  : 254.44 KiB
Packager        : Andreas Radke <andyrtr@archlinux.org>
Build Date      : Mon 01 Feb 2021 08:41:27
Install Date    : Thu 08 Apr 2021 22:32:29
Install Reason  : Installed as a dependency for another package
Install Script  : No
Validated By    : Signature
[/code]

mohamed-badaoui, 2021-04-11:
Ok.
And the output of 
sudo ./install.sh 
?

hjheins, 2021-04-11:
<code>
# ./install.sh 
Testing interface i2c-10 : failed
Testing interface i2c-11 : sucess
Does your numpad has % and = symbols [N/y]y
Your numpad has % and = symbols.

What is your keyboard layout?
1) Qwerty
2) Azerty
3) Quit
Please enter your choice [1-3]: 1
Copy asus python driver to /usr/bin/asus_touchpad_numpad.py
Add asus touchpad service in /lib/systemd/system/
Asus touchpad service enabled
Asus touchpad service started
</code>

But unfortunately pressing F8 doesn't yield a numeric keypad.

mohamed-badaoui, 2021-04-11:
Even with the touchpad top right switch?

hjheins, 2021-04-11:
that one doesn't work at all.
I had F8 working once, after installing the first time, but never since.

mohamed-badaoui, 2021-04-11:
ok.
run sudo ./uninstall.sh

and then execute from the project directory, directly python script:
sudo python3 ./touchpad_numpad_symbols.py 

what is the output?

hjheins, 2021-04-11:
no output. the prompt goes to the next line (no commandline)

hjheins, 2021-04-11:
wait, now it seems to work: I get digits. However I can not see the digits on the touchpad. 
And I get an error:

Error: I2C bus name doesn't match any bus present!

hjheins, 2021-04-11:
I can not see the digits on the touchpad as it's not lighting up.

hjheins, 2021-04-11:
The symbol on the touchpad however seems to be working for switching between the 2 modes.

mohamed-badaoui, 2021-04-11:
ok.
Does this command light up your touchpad ?

i2ctransfer -f -y 11  w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad

hjheins, 2021-04-11:
Nope, unfortunately not:

# i2ctransfer -f -y 11  w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
Error: Could not open file `/dev/i2c-11' or `/dev/i2c/11': No such file or directory

hjheins, 2021-04-11:
When I first install your app, I get no error when performing this command. However the touchpad is not lighting up

hjheins, 2021-04-11:
ooof wait. it does work... delayed a lot..

mohamed-badaoui, 2021-04-11:
If it does not light up the numpad, probably it is due to a arch linux particularity.
Sorry can't help.

closed 2021-04-11 by mohamed-badaoui

hjheins, 2021-04-11:
no no, it does light up now. with this command. it just responded very late to the command. rebooted to see if that takes the delay out.

hjheins, 2021-04-11:
Ok, I found out when it works and when not: when I install your script in a console, and execute in KDE when this console is open (i can even close the console and open it in a session), the touchpad/numpad works, even with the touchpad button for switching between touchpad and numpad. However when the console is closed, it does not work at all. Also after reboor, it does not work anymore. So something seems session related here...

mohamed-badaoui, 2021-04-11:
Ok.
I don't know how services are configured under arch linux.
On debian like distros, services are managed by systemd.
Probably install.sh need some adaptation to arch linux, specially for the service in order to make it working on the reboot.

hjheins, 2021-04-11:
Thank you, I will check how I could play with this and send you feedback if I have it working.

KairuiLiu, 2021-06-25:
> Thank you, I will check how I could play with this and send you feedback if I have it working.

@mohamed-badaoui 
I meet the same problem alse see
```
I2C bus name doesn't match any bus present!
```
and I try found
```
I: Bus=0018 Vendor=04f3 Product=3134 Version=0100
N: Name="ASUE140A:00 04F3:3134 Mouse"
P: Phys=i2c-ASUE140A:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-10/i2c-ASUE140A:00/0018:04F3:3134.0001/input/input10
U: Uniq=
H: Handlers=event4 mouse0 
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
B: REL=1943
B: MSC=10

I: Bus=0018 Vendor=04f3 Product=3134 Version=0100
N: Name="ASUE140A:00 04F3:3134 Touchpad"
P: Phys=i2c-ASUE140A:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-10/i2c-ASUE140A:00/0018:04F3:3134.0001/input/input11
U: Uniq=
H: Handlers=event5 mouse1 
B: PROP=5
B: EV=1b
B: KEY=e520 10000 0 0 0 0
B: ABS=2e0800000000003
B: MSC=20

I: Bus=0018 Vendor=04f3 Product=3134 Version=0100
N: Name="ASUE140A:00 04F3:3134 Keyboard"
P: Phys=i2c-ASUE140A:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-10/i2c-ASUE140A:00/0018:04F3:3134.0001/input/input12
U: Uniq=
H: Handlers=sysrq kbd leds event6 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff800000000007ff febeffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=1f
```
and I try
```
❯ sudo i2ctransfer -f -y 11 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
Error: Sending messages failed: Remote I/O error
❯ sudo i2ctransfer -f -y 12 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
Error: Sending messages failed: Connection timed out
```

it occured every three times i start up

----

- archlinux 
- Kernel: 5.12.12-arch1-1
- ZenBook UX425EA_U4700EA

KairuiLiu, 2021-06-29:
> Thank you, I will check how I could play with this and send you feedback if I have it working.

Hi, it seems that I find the solution

**the hex number value is from py file, you need to copy yours from your py file(becauseof different type of layout)**

1. run the install.sh make sure the numpad works correctly
2. modify  the py file
  ```bash
  vim /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py
  ```
3. add line
```py
print(onCmd)
```
after `offCmd=...`
4. run the py file
```bash
sudo python  /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py <your option like m433ia>
```

5. get the device_id (after i2ctransfer -f -y) from output, for example
```bash
i2ctransfer -f -y 1 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```
the device id is 1
6. find `i2c-<your device_id>`(like `i2c-1`) in `/proc/bus/input/devices`, you may find many lines, then copy all of them(they always begin with `S:`)
7. reboot your laptop until the numpad not work
8. find the lines similar to the lines you copied in `/proc/bus/input/devices`, for me I copied 
```
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-1/i2c-ASUE140A:00/0018:04F3:3134.0001/input/input11
```
and i find `i2c-ASUE140A:00/0018:04F3:3134.0001`
9. then you may get a new device_id from the line, for example `i2c-10` means 10
10. try to run 
```bash
i2ctransfer -f -y  <the device_id you find in step 9> w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```
if it work do step 11
11. modify the py file
```diff
onCmd = "i2ctransfer -f -y " + device_id + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad"
offCmd = "i2ctransfer -f -y " + device_id + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad"

+ onCmd2 = "i2ctransfer -f -y " + "1" + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad"
+ offCmd2 = "i2ctransfer -f -y " + "1" + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad"
- print(onCmd2)
+ 
+ onCmd3 = "i2ctransfer -f -y " + "10" + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad"
+ offCmd3 = "i2ctransfer -f -y " + "10" + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad"
+ 
+ if subprocess.call(onCmd, shell=True)==1:
+         onCmd = onCmd2
+         offCmd = offCmd2
+         if subprocess.call(onCmd, shell=True)==1:
+                 onCmd = onCmd3
+                 offCmd = offCmd3
+                 subprocess.call(onCmd, shell=True)
```
here "1" is the device_id when it run correctly , and 10 is the device_id not run correctly
12. try to restart the service
```
systemctl restart asus_touchpad_numpad.service 
```
13. if the numpad light when the service start it means running, so another tip is you need to touch twice for the first time you want to stop the light (1st time you touch it, the numpad light again)

----

another tip for the situation if it run correctly when you restart the service but do not run currectly when start up:
change py file
```
tries=5
+ sleep(5)
# Look into the devices file #
while tries > 0:
+     sleep(8)
    keyboard_detected = 0
    touchpad_detected = 0

```
the reason is that the service run before numpad be loaded...

if you don't want to wait for 5 seconds when you restart the service, you can write 2 py file, one just run when start up, another run for after start up, register two services, one is for start up, you can also see the [config of mine](https://github.com/KairuiLiu/dotfile)