mohamed-badaoui/asus-touchpad-numpad-driver #11: Enhancement Request: Additional Device: Asus E410MA

arniotis, 2021-04-13:
Greetings. 

I just got an E410MA, and I was happy to find your project to enable the numpad. This device has the % and = buttons, but when running the code as-is the calibration was a bit off, because there is an extra "dead" row on the top over the 7-8-9 keys.

In short, I was able to fix this with the following edits. As you see it is just a matter of adding an extra dead row on the top - I also expanded the "ON/OFF" button area in order to match the are covered by the symbol on the display.

Many thanks for making this driver available.

Michael

@@ -190,3 +190,3 @@ while True:
             e.value == 1 and 
-            (x > 0.95 * maxx) and (y < 0.05 * maxy)
+            (x > 0.90 * maxx) and (y < 0.20 * maxy)
         ):
@@ -207,4 +207,8 @@ while True:
             try:
+                # zero (empty) row
+                if y < 0.2 * maxy:
+                    # space is not enabled, so this intentionally does nothing
+                    value = EV_KEY.KEY_SPACE
                 # first row
-                if y < 0.25 * maxy:
+                elif y < 0.4 * maxy:
                     if x < 0.2 * maxx:
@@ -220,3 +224,3 @@ while True:
                 # second row
-                elif y < 0.5 * maxy:
+                elif y < 0.6 * maxy:
                     if x < 0.2 * maxx:
@@ -232,3 +236,3 @@ while True:
                 # third row
-                elif y < 0.75 * maxy:
+                elif y < 0.8 * maxy:
                     if x < 0.2 * maxx:

mohamed-badaoui, 2021-04-14:
Hi,

Thank you for your contribution.
I just checked the touchpad layout of your model and it seems a little different from the anothers i ve seen until now.

yours is [with no bottom margin](https://media.materiel.net/r900/products/MN0005773638_1.jpg)

but anothers have margin on top and on bottom too, as you can see [for that model for example](https://github.com/mohamed-badaoui/ux433-touchpad-numpad/blob/main/Asus-VivoBook-M433IA.jpg)
 
So what we should do to be compliant with all models, is to identifiy all touchpad layouts first.
Do not know where we could find such a list.

arniotis, 2021-04-15:
Hello again

I think this can be automated using the `dmidecode | grep "Product Name" ` command.  On my laptop it reports 

```
Product Name: VivoBook_ASUS Laptop E410MA_E410MA
Product Name: E410MA
```
Hopefully anyone reading this with an ASUS can post their own results and confirm the type of numpad they have. Once you know a few popular models, you can change the installer to automate the process (at least as to the type of laptop) and have the installer print "If your laptop is not recognised, please report it below "https://github.com/mohamed-badaoui/ux433-touchpad-numpad/issues/11" or something like that...

I'll be happy to help in any way possible

Cheers,

Michael