mohamed-badaoui/asus-touchpad-numpad-driver #12: German keyboard

adricontre, 2021-04-15:
Hi Mohamed! 
Thank you very much for your time and the wonderful solution.
Unfortunately I have a German keyboard and had to change all the keys, except the / and * that I can't get done.
It seems the installation does not accept 'SHIFT_KEY_RIGHTBRACE': '*' (german) the Shift configuration, and for example: 'KEY_KPSLASH': '/',  'KEY_KPASTERISK': '*', (DE) also doesnt work.
Do you know how to find exactly what kind or KEY is using my keyboard or how to use SHIFT?

Thank you again for your time and effort.

mohamed-badaoui, 2021-04-15:
Hi,

In order to try to help you I recommend you to ask first your question in this thread discussion.
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1810183

In this thread, ask your question to " Dennis (bitparade) " (do a CTRL+F search) because for him, it had worked with a german keyboard.

Personally I only did tests on Azerty&Qwerty keyboards but I just switch my keyboard  to Qwertz, it seems to be working.

closed 2021-04-15 by mohamed-badaoui

adricontre, 2021-04-15:
Thank you very much! lets see if I solve it :)