mohamed-badaoui/asus-touchpad-numpad-driver #13: ux363ea

vb42ch, 2021-04-15:
new value of mesh for asus ux363ea

xx = [0.232,0.410,0.588,0.768]
yy = [0.343,0.553,0.763]
            
            try:
                # first row
                if y < yy[0] * maxy:
                    if x < xx[0] * maxx:
                        value = EV_KEY.KEY_KP7
                    elif x < xx[1] * maxx:
                        value = EV_KEY.KEY_KP8
                    elif x < xx[2] * maxx:
                        value = EV_KEY.KEY_KP9
                    elif x < xx[3] * maxx:
                        value = EV_KEY.KEY_KPSLASH
                    else:
                        value = EV_KEY.KEY_BACKSPACE
                # second row
                elif y < yy[1] * maxy:
                    if x < xx[0] * maxx:
                        value = EV_KEY.KEY_KP4
                    elif x < xx[1] * maxx:
                        value = EV_KEY.KEY_KP5
                    elif x < xx[2] * maxx:
                        value = EV_KEY.KEY_KP6
                    elif x < xx[3] * maxx:
                        value = EV_KEY.KEY_KPASTERISK
                    else:
                        value = EV_KEY.KEY_BACKSPACE
                # third row
                elif y < yy[2] * maxy:
                    if x < xx[0] * maxx:
                        value = EV_KEY.KEY_KP1
                    elif x < xx[1] * maxx:
                        value = EV_KEY.KEY_KP2
                    elif x < xx[2] * maxx:
                        value = EV_KEY.KEY_KP3
                    elif x < xx[3] * maxx:
                        value = EV_KEY.KEY_KPMINUS
                    else:
                        value = EV_KEY.KEY_5
                # last row
                else :
                    if x < xx[0] * maxx:
                        value = EV_KEY.KEY_KP0
                    elif x < xx[1] * maxx:
                        value = EV_KEY.KEY_KPDOT
                    elif x < xx[2] * maxx:
                        value = EV_KEY.KEY_KPENTER
                    elif x < xx[3] * maxx:
                        value = EV_KEY.KEY_KPPLUS
                    else:
                        value = EV_KEY.KEY_KPEQUAL

mohamed-badaoui, 2021-04-15:
Thank you very much.
How did you find this exact values for your model?

vb42ch, 2021-04-15:
I measured)
calculations are approximate but better than standard. In this laptop model, the touchpad has a nonlinear grid.
[image]

mohamed-badaoui, 2021-04-15:
Ok cool! Thank you very much.
Do you think that we could use that values for all touchpad with symbols?

vb42ch, 2021-04-15:
I know for sure that for 13 inches ux363ea and ux371e as there are the same 138.5x70 mm touchpads.
Judging by the pictures of other laptops, there are other sizes and proportions of the grid.

mohamed-badaoui, 2021-04-15:
Ok so for 13'' asus models (UX3xxx), it could be right but not suitable for UX4xxx models. 
My touchpad on UX433 (14") is 99x51 mm, counting only the sensitives areas of the touchpad, ie the area limited by enlighted lines.
The whole touchpad is 105x62mm but the borders are not sensitives.

slemal, 2021-07-22:
I have the same laptop and made a [new branch](https://github.com/slemal/asus-touchpad-numpad-driver/tree/ux363) with adjusted values. Feel free to use it.

closed 2021-09-15 by mohamed-badaoui