mohamed-badaoui/asus-touchpad-numpad-driver #18: Does not work on the Asus Zenbook UM425IA

tobfos2611, 2021-04-19:
I have an Asus Zenbook UM425IA, and when I install it, nothing happens and when I reboot, my system is not able to boot until I uninstall it.

mohamed-badaoui, 2021-04-19:
Hi,

It has been tested on UM433DA but not on UM425IA yet.
Before installing the driver, make some verification first.

What does this command return:
```
sudo i2cdetect -l
```

tobfos2611, 2021-04-23:
The first command returns nothing, and the second command returns `I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input13
U: Uniq=
H: Handlers=kbd event13 rfkill 
B: PROP=0
B: EV=100013
B: KEY=1000000080000 0 0 0 0 181606f00900000 8280027801701000 e000000000000 0
--
I: Bus=0018 Vendor=04f3 Product=3134 Version=0100
N: Name="ASUE140A:00 04F3:3134 Mouse"
P: Phys=i2c-ASUE140A:00
S: Sysfs=/devices/platform/AMDI0010:03/i2c-0/i2c-ASUE140A:00/0018:04F3:3134.0001/input/input14
U: Uniq=
H: Handlers=event4 mouse0 
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=3134 Version=0100
N: Name="ASUE140A:00 04F3:3134 Touchpad"
P: Phys=i2c-ASUE140A:00
S: Sysfs=/devices/platform/AMDI0010:03/i2c-0/i2c-ASUE140A:00/0018:04F3:3134.0001/input/input15
U: Uniq=
H: Handlers=event5 mouse1 
B: PROP=5
B: EV=1b
B: KEY=e520 10000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=3134 Version=0100
N: Name="ASUE140A:00 04F3:3134 Keyboard"
P: Phys=i2c-ASUE140A:00
S: Sysfs=/devices/platform/AMDI0010:03/i2c-0/i2c-ASUE140A:00/0018:04F3:3134.0001/input/input16
U: Uniq=
H: Handlers=sysrq kbd leds event6 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff800000000007ff febeffdfffefffff fffffffffffffffe`

tobfos2611, 2021-04-23:
The second one now outputs
`i2c-3	smbus     	SMBus PIIX4 adapter port 1 at 0b20	SMBus adapter
i2c-10	i2c       	AMDGPU DM aux hw bus 3          	I2C adapter
i2c-1	smbus     	SMBus PIIX4 adapter port 0 at 0b00	SMBus adapter
i2c-8	i2c       	AMDGPU DM aux hw bus 1          	I2C adapter
i2c-6	i2c       	AMDGPU DM i2c hw bus 2          	I2C adapter
i2c-4	i2c       	AMDGPU DM i2c hw bus 0          	I2C adapter
i2c-2	smbus     	SMBus PIIX4 adapter port 2 at 0b00	SMBus adapter
i2c-0	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-9	i2c       	AMDGPU DM aux hw bus 2          	I2C adapter
i2c-7	i2c       	AMDGPU DM i2c hw bus 3          	I2C adapter
i2c-5	i2c       	AMDGPU DM i2c hw bus 1          	I2C adapter`
after installing it, this time system continues to boot, but says this while booting
`Starting version 248-5-arch
Can't find touchpad, code 0`

haakov, 2021-05-02:
@tobfos2611 
This might help: https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/pull/21

tobfos2611, 2021-05-16:
I got it working, but there is no way to move the mouse under the numberpad when its on or to adjust the brightness with the button in the top left.

haakov, 2021-05-16:
Yep, mine's the same.

tobfos2611, 2021-05-16:
Oh okay.

tobfos2611, 2021-05-16:
But, I'm getting this after rebooting.
[image]

haakov, 2021-05-16:
Can you share your `sudo journalctl -u asus_touchpad_numpad` ?

tobfos2611, 2021-05-16:
[image]

haakov, 2021-05-16:
Oh okay I suppose the error is being logged somewhere else (if at all). I was getting an OSError on boot from the script, but adding the 1-second sleep in my pull request fixed it for me. Maybe you can try increasing to 3 seconds or something

WatchDogsDev, 2021-05-16:
Same issue here with UX425EA.

When I reboot the system, before reboot completes a terminal like black screen comes with error Can't find touchpad.

And when I try to login after reboot, It shows a black screen completely frozen. 

I tries @haakov workaround but no success. (even with 3 seconds delay)

Please fix it.

WatchDogsDev, 2021-05-16:
When I run this on my ZenBook 14 UX425EA:
`sudo python3 touchpad_numpad_symbols.py`
It gives error "Can't find touchpad, code 0".

But when I run
`sudo python3 touchpad_numpad_nosymbols.py `
It works fine but the numpad has no light and is completely dark.
When I reboot the system, I can not login without removing the service file. (Black screen & freeze as I said before)

mohamed-badaoui, 2021-05-17:
Hi,

Did you try to follow the installation commendation of the user Logix given on launchpad:

WatchDogsDev, 2021-05-17:
I tried the post you gave but no difference.

Should I uninstall any pip3 module ?

mohamed-badaoui, 2021-05-17:
Yes if you did the installation with your system management package, you can pip uninstall the module:

```
pip3 uninstall libevdev
```

What these commands return for your laptop (UM425?)
```
sudo i2cdetect -l
grep -B1 -A5 -i -E "(elan|asu)" /proc/bus/input/devices
```

Disable this installed driver:
```
sudo systemctl stop asus_touchpad_numpad
sudo systemctl disable asus_touchpad_numpad
```

And launch it manually:
```
sudo python3 touchpad_numpad_symbols.py
```

Does this work, do you have some displayed errors?

WatchDogsDev, 2021-05-17:
**i2cdetect -l**
```
i2c-3	i2c       	i915 gmbus dpa                  	I2C adapter
i2c-10	i2c       	i915 gmbus tc5                  	I2C adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-8	i2c       	i915 gmbus tc3                  	I2C adapter
i2c-15	i2c       	AUX F/port F                    	I2C adapter
i2c-6	i2c       	i915 gmbus tc1                  	I2C adapter
i2c-13	i2c       	AUX D/port D                    	I2C adapter
i2c-4	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-11	i2c       	i915 gmbus tc6                  	I2C adapter
i2c-2	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-0	smbus     	SMBus I801 adapter at efa0      	SMBus adapter
i2c-9	i2c       	i915 gmbus tc4                  	I2C adapter
i2c-16	i2c       	AUX G/port G                    	I2C adapter
i2c-7	i2c       	i915 gmbus tc2                  	I2C adapter
i2c-14	i2c       	AUX E/port E                    	I2C adapter
i2c-5	i2c       	i915 gmbus dpc                  	I2C adapter
i2c-12	i2c       	AUX A/port A                    	I2C adapter
```

**grep -B1 -A5 -i -E "(elan|asu)" /proc/bus/input/devices**
```
I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input6
U: Uniq=
H: Handlers=rfkill kbd event4 
B: PROP=0
B: EV=100013
B: KEY=1000000080000 0 0 0 0 181606f00900000 8280027801701000 e000000000000 0
--
I: Bus=0018 Vendor=04f3 Product=30f1 Version=0100
N: Name="ELAN1206:00 04F3:30F1 Mouse"
P: Phys=i2c-ELAN1206:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ELAN1206:00/0018:04F3:30F1.0001/input/input7
U: Uniq=
H: Handlers=mouse0 event5 
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=30f1 Version=0100
N: Name="ELAN1206:00 04F3:30F1 Touchpad"
P: Phys=i2c-ELAN1206:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ELAN1206:00/0018:04F3:30F1.0001/input/input8
U: Uniq=
H: Handlers=mouse1 event6 
B: PROP=5
B: EV=1b
B: KEY=e520 10000 0 0 0 0
```

**sudo python3 touchpad_numpad_symbols.py**
`Can't find touchpad, code 0`

WatchDogsDev, 2021-05-17:
My laptop model is **"Asus ZenBook 14 UX425EA"**

mohamed-badaoui, 2021-05-17:
Ok thanks.
I see that both touchpad models (with and without symbols)  can have ASUE or ELAN keywords on input devices file.
So I updated both python script to match these 2 keywords.

Should be ok by now, just re download the last version and reinstall the project.

Can you confirm?

WatchDogsDev, 2021-05-17:
Good job man. The script is now running without any errors. (Service is also active)

There is no more black screen and freeze on startup login.

I can switch to numpad and buttons work correctly. but there is no light on numpad and I can't see anything.

mohamed-badaoui, 2021-05-17:
In your device file, your device id is 2.

So can you succeed to turn on your numpad with this command and this device id:
```
sudo i2ctransfer -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```

to turn off, It is:
```
i2ctransfer -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad
```

mohamed-badaoui, 2021-05-17:
Please for your model UX425EA, consider to log issue informations at:
https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/22

This issue concerns the  Asus Zenbook UM425IA model

Thanks.

tobfos2611, 2021-05-20:
No this is not fixed yet, I am still unable to use the numberpad after rebooting. Changing the sleep time to 3 seconds did not fix it for me.

mohamed-badaoui, 2021-05-20:
In your device file, your device id seems to be 0

So can you succeed to turn on your numpad with this command and this device id?
```
sudo i2ctransfer -f -y 0 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```

to turn off, It is:
```
sudo i2ctransfer -f -y 0 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad
```

As I said in other threads, if these commands failed, this driver cannot work on this model

tobfos2611, 2021-05-21:
After doing the first one, the lighting on the trackpad turned on, but it wan't working. Then after the second command, the lights turned off and the trackpad worked as normal again.
I'll close this ticket if there is nothing that can be done about this.

mohamed-badaoui, 2021-05-21:
these commands just allow the turning on/off the numpad lights.
it's a good point if that works.
try to run the python script manually and see if it works.

tobfos2611, 2021-05-21:
It works when I do that.

mohamed-badaoui, 2021-05-21:
oh weird oO' 
so it does not work when you run it as a service and you reboot your laptop?
which OS do you use ?

mohamed-badaoui, 2021-05-21:
> I got it working, but there is no way to move the mouse under the numberpad when its on or to adjust the brightness with the button in the top left.

When the numpad is activated, the touchpad is disabled and only the top right area works.

tobfos2611, 2021-05-21:
When I reboot my laptop, it stops working. I'm running Arch Linux with the latest `linux-lts` kernel.

mohamed-badaoui, 2021-05-24:
@tobfos2611 Probably you should try to contact Luca who has succeed to make it work under arch linux and  who wrote on launchpad:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1810183

> Thank you to all, with this instructions here https://www.linuxuprising.com/2021/05/how-to-get-numpad-embedded-into.html work perfectly on fresh arch.
> one little bug, tray icon numpad (plasma) will be removed only after a double on-off
> 
> Thank you

KairuiLiu, 2021-06-29:
Hi, it seems that I find the solution for archlinux

**the hex number value is from py file, you need to copy yours from your py file(becauseof different type of layout)**

1. run the install.sh make sure the numpad works correctly
2. modify  the py file
  ```bash
  vim /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py
  ```
3. add line
```py
print(onCmd)
```
after `offCmd=...`
4. run the py file
```bash
sudo python  /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py <your option like m433ia>
```

5. get the device_id (after i2ctransfer -f -y) from output, for example
```bash
i2ctransfer -f -y 1 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```
the device id is 1
6. find `i2c-<your device_id>`(like `i2c-1`) in `/proc/bus/input/devices`, you may find many lines, then copy all of them(they always begin with `S:`)
7. reboot your laptop until the numpad not work
8. find the lines similar to the lines you copied in `/proc/bus/input/devices`, for me I copied 
```
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-1/i2c-ASUE140A:00/0018:04F3:3134.0001/input/input11
```
and i find `i2c-ASUE140A:00/0018:04F3:3134.0001`
9. then you may get a new device_id from the line, for example `i2c-10` means 10
10. try to run 
```bash
i2ctransfer -f -y  <the device_id you find in step 9> w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```
if it work do step 11
11. modify the py file
```diff
onCmd = "i2ctransfer -f -y " + device_id + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad"
offCmd = "i2ctransfer -f -y " + device_id + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad"

+ onCmd2 = "i2ctransfer -f -y " + "1" + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad"
+ offCmd2 = "i2ctransfer -f -y " + "1" + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad"
- print(onCmd2)
+ 
+ onCmd3 = "i2ctransfer -f -y " + "10" + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad"
+ offCmd3 = "i2ctransfer -f -y " + "10" + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad"
+ 
+ if subprocess.call(onCmd, shell=True)==1:
+         onCmd = onCmd2
+         offCmd = offCmd2
+         if subprocess.call(onCmd, shell=True)==1:
+                 onCmd = onCmd3
+                 offCmd = offCmd3
+                 subprocess.call(onCmd, shell=True)
```
here "1" is the device_id when it run correctly , and 10 is the device_id not run correctly
12. try to restart the service
```
systemctl restart asus_touchpad_numpad.service 
```
13. if the numpad light when the service start it means running, so another tip is you need to touch twice for the first time you want to stop the light (1st time you touch it, the numpad light again)

----

another tip for the situation if it run correctly when you restart the service but do not run currectly when start up:
change py file
```
tries=5
+ sleep(5)
# Look into the devices file #
while tries > 0:
+     sleep(8)
    keyboard_detected = 0
    touchpad_detected = 0

```
the reason is that the service run before numpad be loaded...

if you don't want to wait for 5 seconds when you restart the service, you can write 2 py file, one just run when start up, another run for after start up, register two services, one is for start up, you can also see the [config of mine](https://github.com/KairuiLiu/dotfile), see asus* and REC.sh

KairuiLiu, 2021-06-29:
Hi, it seems that I find the solution for archlinux

**the hex number value is from py file, you need to copy yours from your py file(becauseof different type of layout)**

## solution for it not light sometimes

1. run the install.sh make sure the numpad works correctly
2. modify  the py file
  ```bash
  vim /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py
  ```
3. add line
```py
print(onCmd)
```
after `offCmd=...`
4. run the py file
```bash
sudo python  /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py <your option like m433ia>
```

5. get the device_id (after i2ctransfer -f -y) from output, for example
```bash
i2ctransfer -f -y 1 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```
the device id is 1
6. find `i2c-<your device_id>`(like `i2c-1`) in `/proc/bus/input/devices`, you may find many lines, then copy all of them(they always begin with `S:`)
7. reboot your laptop until the numpad not work
8. find the lines similar to the lines you copied in `/proc/bus/input/devices`, for me I copied 
```
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-1/i2c-ASUE140A:00/0018:04F3:3134.0001/input/input11
```
and i find `i2c-ASUE140A:00/0018:04F3:3134.0001`
9. then you may get a new device_id from the line, for example `i2c-10` means 10
10. try to run 
```bash
i2ctransfer -f -y  <the device_id you find in step 9> w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```
if it work do step 11
11. modify the py file
```diff
onCmd = "i2ctransfer -f -y " + device_id + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad"
offCmd = "i2ctransfer -f -y " + device_id + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad"

+ onCmd2 = "i2ctransfer -f -y " + "1" + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad"
+ offCmd2 = "i2ctransfer -f -y " + "1" + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad"
- print(onCmd2)
+ 
+ onCmd3 = "i2ctransfer -f -y " + "10" + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad"
+ offCmd3 = "i2ctransfer -f -y " + "10" + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad"
+ 
+ if subprocess.call(onCmd, shell=True)==1:
+         onCmd = onCmd2
+         offCmd = offCmd2
+         if subprocess.call(onCmd, shell=True)==1:
+                 onCmd = onCmd3
+                 offCmd = offCmd3
+                 subprocess.call(onCmd, shell=True)
```
here "1" is the device_id when it run correctly , and 10 is the device_id not run correctly
12. try to restart the service
```
systemctl restart asus_touchpad_numpad.service 
```
13. if the numpad light when the service start it means running, so another tip is you need to touch twice for the first time you want to stop the light (1st time you touch it, the numpad light again)

----

## solution for it can't work when start up

another tip for the situation if it run correctly when you restart the service but do not run currectly when start up:
change py file
```
tries=5
+ sleep(5)
# Look into the devices file #
while tries > 0:
+     sleep(8)
    keyboard_detected = 0
    touchpad_detected = 0

```
the reason is that the service run before numpad be loaded...

if you don't want to wait for 5 seconds when you restart the service, you can write 2 py file, one just run when start up, another run for after start up, register two services, one is for start up, you can also see the [config of mine](https://github.com/KairuiLiu/dotfile), see asus* and REC.sh

closed 2021-07-10 by mohamed-badaoui