mohamed-badaoui/asus-touchpad-numpad-driver #19: ux393e backlight

AlfRedOnFire, 2021-04-19:
Hi!

I've an issue with asus ux393e.

Your script is works correctly on my mashine, but touchpad backlight is not working.
Do you have some ideas how to fix it?

mohamed-badaoui, 2021-04-19:
Hi,

It has been tested on your model, so it should work.
what does this command return:

```
grep -B1 -A5 -i -E "(elan|asu)" /proc/bus/
```

AlfRedOnFire, 2021-04-19:
grep: /proc/bus/: Is a directory 

¯\_(ツ)_/¯

AlfRedOnFire, 2021-04-19:
> grep: /proc/bus/: Is a directory
> 
> ¯_(ツ)_/¯

AlfRedOnFire, 2021-04-19:
Sorry missclick :)

mohamed-badaoui, 2021-04-19:
I updated the command, this was truncated :/

AlfRedOnFire, 2021-04-19:
```
I: Bus=0018 Vendor=04f3 Product=2d57 Version=0100
N: Name="ELAN9008:00 04F3:2D57"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-2/i2c-ELAN9008:00/0018:04F3:2D57.0001/input/input12
U: Uniq=
H: Handlers=mouse0 event4 
B: PROP=2
B: EV=1b
B: KEY=400 0 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=2d57 Version=0100
N: Name="ELAN9008:00 04F3:2D57 UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-2/i2c-ELAN9008:00/0018:04F3:2D57.0001/input/input13
U: Uniq=
H: Handlers=event5 
B: PROP=0
B: EV=9
B: ABS=10000000000
--
I: Bus=0018 Vendor=04f3 Product=2d57 Version=0100
N: Name="ELAN9008:00 04F3:2D57 UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-2/i2c-ELAN9008:00/0018:04F3:2D57.0001/input/input14
U: Uniq=
H: Handlers=event6 
B: PROP=0
B: EV=100001

I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input16
U: Uniq=
H: Handlers=rfkill kbd event7 
B: PROP=0
B: EV=100013
B: KEY=1000000080000 0 0 0 0 181606f00900000 8280027801701000 e000000000000 0
--
I: Bus=0018 Vendor=04f3 Product=3134 Version=0100
N: Name="ASUE140A:00 04F3:3134 Mouse"
P: Phys=i2c-ASUE140A:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-11/i2c-ASUE140A:00/0018:04F3:3134.0002/input/input17
U: Uniq=
H: Handlers=mouse1 event8 
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=3134 Version=0100
N: Name="ASUE140A:00 04F3:3134 Touchpad"
P: Phys=i2c-ASUE140A:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-11/i2c-ASUE140A:00/0018:04F3:3134.0002/input/input18
U: Uniq=
H: Handlers=mouse2 event9 
B: PROP=5
B: EV=1b
B: KEY=e520 10000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=3134 Version=0100
N: Name="ASUE140A:00 04F3:3134 Keyboard"
P: Phys=i2c-ASUE140A:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-11/i2c-ASUE140A:00/0018:04F3:3134.0002/input/input19
U: Uniq=
H: Handlers=sysrq kbd event11 leds 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff800000000007ff febeffdfffefffff fffffffffffffffe
```

AlfRedOnFire, 2021-04-19:
Tested again with another OS (previous - Ubuntu classic 20.04, now Ubuntu MATE 20.04 LTS)
Same problem - numberkeys and switching from touchpad to numpad is working, but numpad backlight is not.

Example of my touchpad is on the image below

[image]

mohamed-badaoui, 2021-04-19:
ok.
Try to turn your numpad manually with this commands:
turn on:
```
sudo i2ctransfer -f -y 9 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad"
```
turn off:
```
sudo i2ctransfer -f -y 9  w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad"
```

Does this work properly?

AlfRedOnFire, 2021-04-19:
```Error: Sending messages failed: No such device or address
```

AlfRedOnFire, 2021-04-19:
```Error: Sending messages failed: No such device or address```

mohamed-badaoui, 2021-04-19:
Oh sorry it was not 9 but 11 on yout machine. I updated the command above.

AlfRedOnFire, 2021-04-19:
Yes, if i use this commands - backlight turns off\on. But in your script its not working

And i found another issue - turn on/off sensor (right upper corner of touchpad) not working......or very rarely working.
now i'm using FN+F8 combination to switch touchpad/numpad mode.

mohamed-badaoui, 2021-04-19:
I'm not sure to understand your reply.
You said first the script is not working.
And then you continue by saying that the touchpad switch is very rarely working and the FN+F8 combination works.

So do I have to understand that the script works finally?

> Yes, if i use this commands - backlight turns off\on. But in your script its not working
> 
> And i found another issue - turn on/off sensor (right upper corner of touchpad) not working......or very rarely working.
> now i'm using FN+F8 combination to switch touchpad/numpad mode.

mohamed-badaoui, 2021-04-19:
Try to contact @angryziber  ( https://github.com/angryziber ) 
He has tested this script on this model and it was working for him.

Sorry but there is nothing more I can do to help you.

closed 2021-04-19 by mohamed-badaoui

AlfRedOnFire, 2021-04-19:
1. Your script is working, but on\off sensor on the touchpad is NOT. I need to use FN+F8 instead of this button
2. Backlight of touchpad is not working by your script, but it working when i enter your commands MANUALY

AlfRedOnFire, 2021-04-19:
Nevermind, i found how to fix this problem. It was wrong coords for on\off sensor for touchpad in your script.

THANK YOU VERY MUCH!❤️

angryziber, 2021-04-19:
Can you share the coordinates?
The default ones work for me, bit they are quite far in the corner, no exactly where the icon is drawn

AlfRedOnFire, 2021-04-21:
@angryziber 
`# Check if numlock was hit #
        if (
            e.matches(EV_KEY.BTN_TOOL_FINGER) and 
            e.value == 1 and 
            (x > 0.92 * maxx) and (y < 0.13 * maxy)
        ):`
        
Edit this piece of code like me :) (starts from line 187 in touchpad_numpad_symbols.py)

And fix for backlight:
`onCmd = "i2ctransfer -f -y 11 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad"
offCmd = "i2ctransfer -f -y 11 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad"`
(starts from line 11 in touchpad_numpad_symbols.py)

angryziber, 2021-04-21:
> onCmd = "i2ctransfer -f -y 11 ***@***.*** 0x05 0x00 0x3d 0x03 0x06 0x00 0x07

Yep, I've seen this code already, but what exactly did you have to change?
I guess you moved the triggering point a bit away from the right-top corner?

I don't have problems with backlight, or what do you needed to fix?

AlfRedOnFire, 2021-04-21:
Yes, i moved triggering point from the corner to the right place and now its working when i press to the on\off trigger point directly.
[image]

What about backlight - somehow a mohamed's script dont activate the backlight on my ZenBook S UX393EA and i need to insert a device_id variable manualy to fix this. And i wrote this fix for someone who will have the same problem :)

ZdenoF, 2021-12-02:
Hi guys,

I'm pretty new to linux (finally decided to get rid of windows after many years). I'm runnig on Ubuntu 21.10 now...

I have Asus Zenbook 14 UX431DA which originally came with standard touchpad with fingerprint
sensor. So I decided to buy numpad mod hardware piece for my notebook. Succesfully installed yesterday, also did install driver package. Everything is working correctly but without backlight.
When I press the numlock button on it, I can type the numbers and symbols, but I'm not able to toggle the brightness (backlight), so I don't see anything on the touchpad.
As I am working a lot in excel I really need to get a numpad working properly. So I hope that somebody could help me with this.
Thanks, Zee