mohamed-badaoui/asus-touchpad-numpad-driver #22: UX425EA no light

Datalux, 2021-05-09:
Model: UX425EA
OS: Linux Mint 20.1

After two reboots the brightness of the numpad is completally dark. How can I solve this problem?

mohamed-badaoui, 2021-05-17:
I see that both touchpad models (with and without symbols) can have ASUE or ELAN keywords on input devices file.
So I updated both python scripts to match these 2 keywords.

Should be ok by now, just re download the last version and reinstall the project.

Can you confirm?

WatchDogsDev, 2021-05-17:
> In your device file, your device id is 2.
> 
> So can you succeed to turn on your numpad with this command and this device id:
> 
> ```
> sudo i2ctransfer -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
> ```
> 
> to turn off, It is:
> 
> ```
> sudo i2ctransfer -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad
> ```

Tried both commands but not working. Without any errors or messages.

mohamed-badaoui, 2021-05-17:
This user had confirmed that this model works with this driver on launchpad:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1810183
> Lux Libertas (luxlibertas)             wrote             on 2021-04-15: | #241
> Works perfectly well on UX425EA!
> Thanks!!!

So I suggest you to try to contact him.

But if these command doesnt turn on/off your touchpad, I'm afraid of that your model coultd be not compatible with this driver at all :(

WatchDogsDev, 2021-05-17:
Yes the problem is strange.

Ubuntu 20.04

mohamed-badaoui, 2021-05-17:
This driver just identify your touchpad device id and run these commands to switch your touchpad to numpad.
So if these commands do not work natively, the driver can not neither.

If I do not have positive feedbacks, I will probably remove this model from the compatible list on the README file.

closed 2021-05-17 by mohamed-badaoui

WatchDogsDev, 2021-05-17:
I think I found the problem. My touchpad is ELAN but I think your cmd for i2ctransfer is for ASUE models.

Can you give the link of technical manual ? 

I couldn't find it on the internet. Maybe I can help you fix it.

mohamed-badaoui, 2021-05-17:
The i2ctransfer is the same one for ELAN/ASUE models.
The only things that changes between laptop is the device id.
So when you run theses commands, if they don't work in a console mode, the driver cannot work.

There is no thehnical manual. Asus team refused to give any help.
It was only community people who guess by retro ingeneering methods the parameters of ic2transfer.

Have a look at the whole story on launchpad thread:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1810183

WatchDogsDev, 2021-05-17:
What a shame on asus.

I'm gonna decompile their windows driver. I will post here if I got it working.

batouei, 2021-05-29:
> I think I found the problem. My touchpad is ELAN but I think your cmd for i2ctransfer is for ASUE models.
> 
> Can you give the link of technical manual ?
> 
> I couldn't find it on the internet. Maybe I can help you fix it.

Hi. if you found a way to resolve this problem please tell me. I also have a zenbook ux425 ea.

ZIMkaRU, 2022-03-26:
any news on it?