mohamed-badaoui/asus-touchpad-numpad-driver #24: OS boot error on ASUS ROG STRIX 15 (ubuntu 20.04, 5.8.0)

gitrequests, 2021-05-17:
After install on ASUS ROG STRIX 15 (G532LWS) Ubuntu does not boot to login screen. Can boot only console mode.

`/proc/bus/input/devices`

```bash
I: Bus=0018 Vendor=04f3 Product=315e Version=0100
N: Name="ELAN1405:00 04F3:315E Touchpad"
P: Phys=i2c-ELAN1405:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-3/i2c-ELAN1405:00/0018:04F3:315E.0001/input/input20
U: Uniq=
H: Handlers=mouse2 event17 
B: PROP=1
B: EV=1b
B: KEY=e520 30000 0 0 0 0
B: ABS=2e0800000000003
B: MSC=20
```

If necessary, I can provide additional information.

mohamed-badaoui, 2021-05-17:
Hi,

Did you try to follow the installation recommendations of the user Logix given on launchpad:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1810183

He wrote a post at:
https://www.linuxuprising.com/2021/05/how-to-get-numpad-embedded-into.html?m=1

I updated the README to avoid using Python PIP for the installation.

gitrequests, 2021-05-17:
Yes, I used exactly this instruction, with python3-libevdev, not PIP library.

mohamed-badaoui, 2021-05-17:
Disable this installed driver:
```
sudo systemctl stop asus_touchpad_numpad
sudo systemctl disable asus_touchpad_numpad
```

And launch it manually:
```
sudo python3 touchpad_numpad_nosymbols.py
```

Does this work, do you have some displayed errors?

gitrequests, 2021-05-17:
I already disable driver and boot up to the system. If I launch script manually, there is nothing happens. Touchpad works, but switch to Numpad not working.

mohamed-badaoui, 2021-05-17:
Can you succeed to turn on your numpad with this command:
```
sudo i2ctransfer -f -y 3 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```

to turn off, It is:
```
i2ctransfer -f -y 3 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad
```

If these commands don't work, else i'm of afraid that your model is not compatible with this driver at all.

gitrequests, 2021-05-17:
Yes, led on touchpad is turned on (but it seems that the buttons are not being pressed)

mohamed-badaoui, 2021-05-17:
These commands are the ones used by the script to turn on/off your touchpad light.
Did you try with fn+F8 or F8 after running the python script?

gitrequests, 2021-05-17:
With fn+F8 or F8 there is nothing happens

short video:
https://drive.google.com/file/d/1palvZ2y4VZSJ_SE0uOJXMvDbE0MlKN3Y/view?usp=sharing

mohamed-badaoui, 2021-05-17:
Ok good idea for the video.

So to sum up, your numpad works, I mean that you can print numbers when you hit the right touchpas areas when numpad is on.
But the light are always off?

So I guess the only reason is that the device id is not correctly found.

Can you share the complete result of that command:
```  
grep -B1 -A5 -i -E "(elan|asu)" /proc/bus/input/devices
 ```
 
 Perhaps you have a conflict name of touchpad devices.

gitrequests, 2021-05-17:
Yes, I trying to touch top right area (and for 1 second, as written on the sticker), no result.

`grep -B1 -A5 -i -E "(elan|asu)" /proc/bus/input/devices`

```bash
I: Bus=0019 Vendor=1043 Product=0000 Version=0000
N: Name="Asus Wireless Radio Control"
P: Phys=asus-wireless/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/ATK4002:00/input/input8
U: Uniq=
H: Handlers=rfkill kbd event8 
B: PROP=0
B: EV=3
--
I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input9
U: Uniq=
H: Handlers=rfkill kbd event5 
B: PROP=0
B: EV=100013
B: KEY=1000000080000 0 0 0 0 181606f00900000 8280027801701000 e000000000000 0
--
I: Bus=0018 Vendor=04f3 Product=315e Version=0100
N: Name="ELAN1405:00 04F3:315E Mouse"
P: Phys=i2c-ELAN1405:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-3/i2c-ELAN1405:00/0018:04F3:315E.0001/input/input19
U: Uniq=
H: Handlers=mouse1 event16 
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=315e Version=0100
N: Name="ELAN1405:00 04F3:315E Touchpad"
P: Phys=i2c-ELAN1405:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-3/i2c-ELAN1405:00/0018:04F3:315E.0001/input/input20
U: Uniq=
H: Handlers=mouse2 event17 
B: PROP=1
B: EV=1b
B: KEY=e520 30000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=315e Version=0100
N: Name="ELAN1405:00 04F3:315E Keyboard"
P: Phys=i2c-ELAN1405:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-3/i2c-ELAN1405:00/0018:04F3:315E.0001/input/input21
U: Uniq=
H: Handlers=sysrq kbd event18 leds 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff800000000007ff febeffdfffefffff fffffffffffffffe
--
I: Bus=0003 Vendor=0b05 Product=1866 Version=0110
N: Name="ASUSTeK Computer Inc. N-KEY Device"
P: Phys=usb-0000:00:14.0-8/input0
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:0B05:1866.0005/input/input24
U: Uniq=
H: Handlers=sysrq kbd event21 leds 
B: PROP=0
--
I: Bus=0003 Vendor=0b05 Product=1866 Version=0110
N: Name="ASUSTeK Computer Inc. N-KEY Device"
P: Phys=usb-0000:00:14.0-8/input1
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:0B05:1866.0006/input/input25
U: Uniq=
H: Handlers=sysrq kbd event22 leds 
B: PROP=0
--
I: Bus=0003 Vendor=0b05 Product=1866 Version=0110
N: Name="ASUSTeK Computer Inc. N-KEY Device"
P: Phys=usb-0000:00:14.0-8/input2
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:0B05:1866.0007/input/input26
U: Uniq=
H: Handlers=event23 
B: PROP=0
--
I: Bus=0003 Vendor=0b05 Product=1866 Version=0110
N: Name="ASUSTeK Computer Inc. N-KEY Device Consumer Control"
P: Phys=usb-0000:00:14.0-8/input2
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:0B05:1866.0007/input/input27
U: Uniq=
H: Handlers=kbd event24 
B: PROP=0
--
I: Bus=0003 Vendor=0b05 Product=1866 Version=0110
N: Name="ASUSTeK Computer Inc. N-KEY Device"
P: Phys=usb-0000:00:14.0-8/input2
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:0B05:1866.0007/input/input28
U: Uniq=
H: Handlers=event25 
B: PROP=0
--
I: Bus=0003 Vendor=0b05 Product=1866 Version=0110
N: Name="ASUSTeK Computer Inc. N-KEY Device"
P: Phys=usb-0000:00:14.0-8/input2
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:0B05:1866.0007/input/input29
U: Uniq=
H: Handlers=event26 
B: PROP=0
```

mohamed-badaoui, 2021-05-17:
Really weird.
I suggest you to add these print log in the ptyhon script right after onCmd and offCmd lines (around L.111/112)

```
print(onCmd)
print(offCmd)
```

Run it manually and check if it is the expected commands.

gitrequests, 2021-05-17:
`sudo python3 touchpad_numpad_nosymbols.py`

```bash
i2ctransfer -f -y 3 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
i2ctransfer -f -y 3 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad
```

mohamed-badaoui, 2021-05-17:
Ok more and more strange oO'

Sorry but I don't see what could be wrong on your system.

closed 2021-05-17 by mohamed-badaoui