mohamed-badaoui/asus-touchpad-numpad-driver #26: Small issues with Asus UX425EA

nicosensei, 2021-05-26:
Hi,

On my Asus UX425EA, using the top right corner of the touchpad to activate/deactivate the numeric pad does not work: it works once after installing the driver, then never again.

Also sporadically the touchpad is deactivated when switching off the numpad.

How can I help debugging?

mohamed-badaoui, 2021-05-26:
Hi,

I suggest you to post on launchpad forum:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1810183

This model has been confirmed as compatible with this driver.
Check if you have errors in your system logs.
(It depends on your linux distro)

Also uninstall the service and run it manually, so you can check what goes wrong.

nicosensei, 2021-05-26:
Actually after a bit of testing it does work. But you have to touch really
close to the corner, if you press the pictogram it does not work.

No big deal, though.

Cheers !

mohamed-badaoui, 2021-05-26:
ok great!

closed 2021-05-26 by mohamed-badaoui