mohamed-badaoui/asus-touchpad-numpad-driver #28: Linux unbootable due to drivers

Klintyfye, 2021-05-29:
Hi. I installed the software and after restarting my computer it wont boot fully. : 

^[;lkjk;^@^CTraceback (most recent call last):
    File "/usr/bin/asus_touchpad_numpad.py", line 262, in <module>
        Sleep(0.1)
KeyboardInterrupt
[    64.573405] ucsi_acpi USBC000:00 PPM init failed (-110)

mohamed-badaoui, 2021-05-29:
Hi,

First uninstall the driver:

sudo ./uninstall.sh

and then execute manually from the project directory, directly python script:
sudo python3 ./touchpad_numpad_symbols.py

And check if it works before installing it.

Klintyfye, 2021-05-29:
HI,

I managed to uninstall it and get the whole thing booting again. 

I'm running Kali on the UX425E

Frhazz, 2021-05-29:
[image]

I'm stuck on this ucsi_acpi USBC000:00: PPM init failed (-110)

How can I sort this out? Thank you

mohamed-badaoui, 2021-06-09:
Just run unstall.sh in a console mode from driver directory.
```
sudo ./uninstall.sh
```

Frhazz, 2021-09-18:
Hi, sorry but I tried to give another go to the service, I'm still not able to boot and now I can't uninstall the it.. I've a UX425IA running Pop_Os 20.04
[image]

mohamed-badaoui, 2021-09-18:
Hi,

it seems that your i2c module are not deactivatable.

so uninstall the rest, doing:
```
sudo systemctl stop asus_touchpad_numpad
sudo systemctl disable asus_touchpad_numpad
sudo rm -f /lib/systemd/system/asus_touchpad_numpad.service
sudo rm -rf /usr/share/asus_touchpad_numpad-driver/
```

Frhazz, 2021-09-18:
It does work when I start it manually. 

I've disabled it and start it manually once I'm logged in, that's the only solution I found to avoid booting issues. 
I tried to modify the service as mentioned in other issues but with no luck.

mohamed-badaoui, 2021-09-19:
For some distro, there is an issue with the service startup. It does not depend on the python script itself.
At this moment, service fails mainly to start on Mint, Pop!Os, ElementaryOs, and SolusOs.

I have just add an error log file to save service error on startup.
Please to uninstall/install, reboot and check the file
```
cat /var/log/asus_touchpad_numpad-driver/error.log
```

Frhazz, 2021-09-19:
[image]

It says no file of folder of this type

mohamed-badaoui, 2021-09-19:
Did you update your local repo?
Did you run uninstall.sh then install.sh?

It is those scripts which create the log file.

Frhazz, 2021-09-19:
I uninstalled, deleted the folder then pulled a fresh copy from the repo, installed it and reboot. I can redo the manipulation to be sure I didn't miss a part but I don't think I did

mohamed-badaoui, 2021-09-19:
Ok maybe I did something wrong but I do not see what :(

Yes in your case you need to finisj the uninstallation manulaly because i2c module seems to be undeactivatable in your linux OS. 

> I uninstalled, deleted the folder then pulled a fresh copy from the repo, installed it and reboot. I can redo the manipulation to be sure I didn't miss a part but I don't think I did.
> 
> Just to mention again that I can't run sudo ./uninstall.sh, I end up with
> `modprobe: FATAL: Module i2c_dev is builtin. i2c-dev module cannot be removed successfuly..`
> 
> so I do
> `sudo rm -f /lib/systemd/system/asus_touchpad_numpad.service sudo rm -rf /usr/share/asus_touchpad_numpad-driver/`

mohamed-badaoui, 2021-09-19:
can you share this command result
```
cat /etc/systemd/system/asus_touchpad_numpad.service
```

Frhazz, 2021-09-19:
Did the manipulation again but same result :/

Here is the result asked:
`[Unit]
Description=Asus Touchpad to Numpad Handler

[Service]
Type=simple
ExecStart=/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py m433ia 40 > /var/log/asus_touchpad_numpad-driver/error.log 2>&1
StandardInput=tty-force
TimeoutSec=10

[Install]
WantedBy=default.target
`

mohamed-badaoui, 2021-09-19:
Ok my bad :/
It is not the right way to log error by systemd.

The config should be
```
ExecStart=/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py $LAYOUT $PERCENTAGE_KEY
StandardInput=tty-force
StandardOutput=/var/log/asus_touchpad_numpad-driver/error.log
StandardError=/var/log/asus_touchpad_numpad-driver/error.log
```

Repo updated.

Frhazz, 2021-09-19:
Uninstalled and pulled the latest repo but getting the same message saying no such file.

mohamed-badaoui, 2021-09-19:
Ohhhh :(

Does journal contains specific logs entries?

```
sudo journalctl -u asus_touchpad_numpad
```

Frhazz, 2021-09-19:
It just says: Started Asus Touchpad to Numpad Handler.

mohamed-badaoui, 2021-09-19:
Ok thank you for all your feedback.
Sorry but I have no other idea at the moment.

mohamed-badaoui, 2021-09-19:
What you probably could do is to configure the service after user login.
Have a look at:
https://superuser.com/questions/1037466/how-to-start-a-systemd-service-after-user-login-and-stop-it-before-user-logout

Frhazz, 2021-09-19:
I'll leave it as it is for now, not very confortable with all the services and commands, I don't want to screw it up --' 
It's working fine, just have to remember starting the service manually. If someone find a workaround, I'll give it a go in the meantime thank you for this great work, it's nice to be able to use the feature :) and also thank you for take the time to help out.

mohamed-badaoui, 2021-09-19:
Of course I understand. 
Hope someone else could succeed to find why service do not start automatically on  Mint, Pop!Os, ElementaryOs, and SolusOs.

I only use Debian and no really have time to try it with these distros.

mohamed-badaoui, 2021-09-19:
Have a look to the solution suggested by @zenusr https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/68

Frhazz, 2021-09-19:
Didn't work unfortunately :/

EmilCRode, 2021-11-24:
I am currently trying to figure out the boot issue on Pop!OS. Is there maybe a problem with the default location of the service in the home directory which is not reachable on boot?

arivarton, 2021-12-21:
> What you could probably do is configure the service to start after the user logs in. Have a look at: https://superuser.com/questions/1037466/how-to-start-a-systemd-service-after-user-login-and-stop-it-before-user-logout
> 
> the main diffrence is that the service has to be started by
> 
> ```
> systemctl --user enable asus_touchpad_numpad
> ```

I had the same issue on Pop OS and used this as a workaround. But there are several pitfalls mostly concerning permissions for the user so these are the steps I had to take for anyone having the same issue:

Add user to "input" and "i2c" groups. 

Add a udev rule file in /etc/udev/rules.d/ and add this line to it: 
KERNEL=="uinput", MODE="0666"

Then I also had to comment out some lines in the asus_touchpad.service file which were:
StandardInput, StandardOutput, StandardError.
It might be enough to comment out only some of those or change log location from /var/log to something the user has access to, but I didn't care to check this since it is working now.

DevLaTarte, 2022-07-21:
Hello,

I just installed Pop OS 22.04 LTS on my Laptop UX393EA. I looked for a solution about the numpad and installed this.
It work perfectly but I wouldn't be able to boot after installing it.

I just had to comment out the lines "StandardInput, StandardOutput, StandardError" in order to have the service properly starting at boot without blocking.

Thanks @arivarton  for pointing this out.

Also the sleep with ExecStartPre was not doing it whatever the sleep value for me.

And Thanks to all contributors on this drivers, this is really awesome. :fire: