mohamed-badaoui/asus-touchpad-numpad-driver #35: Blocks Bootloader in Pop Os!

ASKabalan, 2021-06-15:
Hello,

This service blocks the boot process for Pop Os!
I haven't tried it on Ubuntu but it seems to work pretty well judging by the comments

Please note that Pop Os uses a different bootLoader than Ubuntu.
There is a work around this (disabling the Service before shutdown and starting it after the boot process) but I think that a solution out of the box is better.
I think an easy solution is to run the main loop in a secondary thread.
What do you think?

vkehayas, 2021-06-15:
I use Pop OS with an ASUS UX333FA and have not faced any problems with booting.
Are you sure that there is not something else causing the problem?

ASKabalan, 2021-06-16:
I have UX362FA and I am positively sure that it is the problem, because in the boot process i see asus-numpad-driver daemon appear and the console freezes.
When I get the chance I will redo the test and share a screenshot.

fboris, 2021-06-19:
my UM425UA also has similar issue. I tried #21. But it didn't help.

Joan-Tondo, 2021-06-19:
I had to modify the service because the gnome doesn't start till I click Alt+F2, command line, startx:
Maybe it works too for you:
`
[Unit]
Description=Asus Touchpad to Numpad Handler
After=graphical.target

[Service]
ExecStart=/usr/bin/python3 /usr/bin/asus_touchpad_numpad.py

[Install]
WantedBy=graphical.target
`

ASKabalan, 2021-06-19:
OK that's a good idea Joan .
Thank you 😁

closed 2021-07-10 by mohamed-badaoui