mohamed-badaoui/asus-touchpad-numpad-driver #46: It activate the numpad on Fn+F8 instead of F8

fmorato, 2021-07-09:
Event though the shortcut is under the F8 key, it should activate on Fn+F8, and not on F8. On devices that have the Fn-Lock, if the Lock is active, pressing F8 will not activate the numpad as would be expected.

slemal, 2021-07-21:
Alternatively, on some layouts, such as the M433IA, the Fn+F10 and Fn+F12 keys are completely useless, so using one of these would be nice.

kawaegle, 2021-07-22:
@fmorato I'm not really sure about the utility of using a key map shortcut to toggle the numpad/touch pad because there is already a "toggle switch" on the track pad and in my opinion it's a better way to use that then key map as you suggest

fmorato, 2021-07-24:
It's not just about utility. If this is the driver for the touchpad, I think it should aim at implementing the original functionality, which includes a toggle on fn+F8. 

About the utility, sometimes I find it easier to toggle with a keystroke than scouting with my finger for the correct spot on the touchpad to press, which often feels like quite a small area.

kawaegle, 2021-07-24:
@fmorato  I'm currently working on the size of the trackpad spot (and I also let the admin the right to modify my version for the fn+F8)

slemal, 2021-07-25:
@fmorato As far as I know, toggle on Fn+F8 isn't part on the original functionality. On my device, Fn+F8 is supposed to cycle through the second screen options, so it would not be convenient to hijack the FN+F8 key. I guess making it customizable would be a lot better.

kawaegle, 2021-07-25:
@slemal  @fmorato soon I hope to be ready to send a PR for changing size of the toggle area but it's on an asus UX425E if someone can tell me if it work on another device it'll be a great help for me and the update. 
In same time I'm also trying to add other keymap support for toggle keymap when installation (need more time for that)

I'm currently unable to test but I hope in august to be able to push and make a PR

thanks

slemal, 2021-07-26:
@kawaegle I tried making something similar [here](https://github.com/slemal/asus-touchpad-numpad-driver). You can change the size of the numlock button, calculator button and offsets by adding a new layout. To test it, you'll also need to update the `install.sh` script. (If someone want to update this script so that it lists the files in `numpad_layout` instead, feel free.)

kawaegle, 2021-07-27:
@slemal How did you set the offset (I mean how did you get a number ?) I use evtest tu get X and Y position and in my case (UX425E) the toggle is in X (3600<x<3900) Y (100<y<230)

and I'm working on the install.sh because my systemd doesn't start it at boot

hope that can help

kawaegle, 2021-07-27:
and also better toggle it on F6 because it show a "touchpad" so I think more logic that key

slemal, 2021-07-27:
I used a ruler to get the offset.

Isn't the F6 key for disabling the touchpad?

kawaegle, 2021-07-27:
oh shift 
I work with i3 so I need to set everything my bad yes under gnome and mate F6 is to disable touchpad 
and F8 is the "monitor manager" maybe try with F12 that the My asus key and my asus doesn't exist on linux

still on it

fmorato, 2021-07-27:
@slemal Indeed, devices have different function keys, so the fn+F8 should be only for devices that have it. You can see from the screenshots in the repo that only UX433FA have the functionality to cycle through the screens, while both the other layouts toggle the numpad on fn+F8.

kawaegle, 2021-07-29:
@slemal I just finish to remove support for fn+f6 key but I improve size of toggle zone and I also finish brightness (but maybe need some little more effort to get value as I want)

closed 2021-09-09 by mohamed-badaoui