mohamed-badaoui/asus-touchpad-numpad-driver #48: ASUS UX325J notebook

garnierj, 2021-07-15:
it works correctly on this model except for the "brightness control" key but that's a detail!
thank you so much

slemal, 2021-07-21:
I guess the command to change the brightness would be something similar to the on command:
`onCmd = "i2ctransfer -f -y " + device_id + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad"`
@mohamed-badaoui do you know which byte sets the brightness?

kawaegle, 2021-07-22:
@slemal @mohamed-badaoui 

```python
onCmd = "i2ctransfer -f -y " + device_id + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad"
offCmd = "i2ctransfer -f -y " + device_id + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x00 0xad"
```

can you explain how did you get that value cause I'm sure I can help using my free time

slemal, 2021-07-25:
It has been mentionned in issue #32, but I couldn't find an explanation on how that command was found.

kawaegle, 2021-07-27:
@mohamed-badaoui @garnierj @slemal as I said on #46 I tried with a little python script to switch all single hexa value in ```w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad``` and the value that set brightness is the penultimate I'm currently trying all value to select 3 value

Hope giving news soon

kawaegle, 2021-07-28:
as I mentioned in my PR #55 I add a python script to test the brightness and already set some value but maybe need more test for select the correct value (and need to be implemented on the python scritp) and I'm sure tomorow I can work all day long on it and maybe send another PR for add officialy the 3 level of bright

Hope that work

kawaegle, 2021-07-29:
Hello guys I just wanted to say that I just finish to add brightness support and little rework the sensitivity for toggle numpad and brightness

I make that clean and I push it

Hope that news make you happy

slemal, 2021-07-29:
I used your script to make some tests and from what I can see, you can get almost any brightness by using a value between 16 and 31, 16 being the brightest and 31 the faintest.

kawaegle, 2021-07-29:
@slemal I'm trying to Get 3 value for distinct bright if you want to give some I'm open to

slemal, 2021-07-29:
I think the values used by windows are 1 (full brightness) and 24 (half brightness). I can't be sure since I need to reboot my computer to compare, but I think that's close enough.
If you want to use a third value, you could use 31 (lowest brightness).

slemal, 2021-07-29:
Dealing with two values can be done in a more elegant way though.

kawaegle, 2021-07-29:
> Dealing with two values can be done in a more elegant way though. 

In more elegant way...
Can you explain better ?

slemal, 2021-07-29:
You can bitwise xor your 'bright' variable to switch rather that cycling through states. It's just a details though. 😅

kawaegle, 2021-07-30:
I'm what we call a newbie but I try my best to get the job done, and I'm on my way to see what's that and if I understand how that work I'll try to upgrade my code

Thanks for the idea (even if I'm not sure about how it work)