mohamed-badaoui/asus-touchpad-numpad-driver #59: TouchPad not highlited (ZenBook UX425EA_UX425EA)

AntonSh23, 2021-08-17:
Hi, i have next laptop
[image]

tried all installation options but i have next issue:
Numpad not highlighted(it just dark like turned off), but if i press right top corner i can write numbers like numpad is working properly.
Can you please direct me where i can find part of code which response for highlight of touch bar?
Tried to find by myself but with no result.
[image]

kawaegle, 2021-08-30:
hello,

Can you show me the output of 

```sh
i2c-detect -l
```

```sh
uname -a
```

```sh
sudo systemctl status asus_touchpad_numpad.service
```

thx

AntonSh23, 2021-09-03:
Hi @kawaegle , thx a lot for reply.
i2cdetect -l
[image]
uname -a
[image]
sudo systemctl status asus_touchpad_numpad.service
[image]

kawaegle, 2021-09-04:
can you try to install my fork I improved the file for this computer (because it's also mine) maybe it's with the i2c command the computer don't understand what to light up

https://github.com/oppaiweeb/asus-touchpad-numpad-driver

I wait for my PR to be accepted

mohamed-badaoui, 2021-09-10:
Hi,

You should try it first manually.
It seems in your case that the right interface is number 2
```
i2c-2	unknown       	Synopsys DesignWare I2C adapter 	I2C adapter
```

So you could try to activate your numpad with
```
sudo i2ctransfer -f -y 2 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad
```

For information, PR has been accepted (Thanks Kawaegle) 
If manually command works, you should uninstall first and refresh your local repository (git pull) before reinstalling

mohamed-badaoui, 2021-09-15:
Generally when the touchpad stay unlighted, it s mainly because i2c interface is not started.
Please check that when you run it manually, i2c interface has been well started.
So before running the script, start i2c interfaces and check it is running.

```
sudo modprobe i2c-dev
sudo i2cdetect -l
```

closed 2021-09-15 by mohamed-badaoui

arturf, 2022-01-30:
Guys just for fun. Some models from UX425EA series don't have Numpad. But driver works fine even without highlight. 
Probably all models have the similar hardware but Asus decided to make some economy on led lights :)
This is my case with a new laptop. Will change it and try again...