mohamed-badaoui/asus-touchpad-numpad-driver #60: Install Script Tries to Place Service File in Wrong Directory for Solus OS

moore-bryan, 2021-08-28:
During install on my Solus 4.3 machine, the script threw an error:
`./install.sh: line 98: /lib/systemd/system/asus_touchpad_numpad.service: No such file or directory
Failed to enable unit: Unit file asus_touchpad_numpad.service does not exist.
Something gone wrong while enabling asus_touchpad_numpad.service`

Changing line 98 from installing the service file in `/lib/systemd/system/` to `/etc/systemd/system/` solved the issue for me.

kawaegle, 2021-08-30:
maybe an error with the different os.

I guess that because i'm only working on ubuntu and on arch never try other distribution but maybe i'm going to add some OS to test 

btw thx for the issue 

(if your issue is done please close it)

moore-bryan, 2021-08-30:
I'm sure it's something distro related; although, I didn't think Arch- or Debian-based systems put systemd services files in /lib... Learned something knew! :-)

My more serious issue is in [this bug](https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/61). Essentially, using this service won't permit X to start. Fun, huh?

mohamed-badaoui, 2021-09-10:
service path fixed by  https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/commit/03b8b2c3e706db9288c58338394a3a26f9f47db4

closed 2021-09-10 by mohamed-badaoui