mohamed-badaoui/asus-touchpad-numpad-driver #61: Install Won't Let X11 Load, No Errors

moore-bryan, 2021-08-28:
After installing on Solus 4.3 and rebooting, there was no GUI and X11 appeared not to have started; additionally, there was no way to start it... It just refused.

Running `sudo systemctl disable asus_touchpad_numpad.service` allows everything to start appropriately.

Any ideas?

kawaegle, 2021-08-30:
hey can you show me the output of 
```sh
sudo systemctl status
``` 
with **and** without the asus_touchpad service? 

thx

moore-bryan, 2021-08-30:
[asus_touchpad_numpad_service-enabled.txt](https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/files/7078352/asus_touchpad_numpad_service-enabled.txt)
[asus_touchpad_numpad_service-disabled.txt](https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/files/7078353/asus_touchpad_numpad_service-disabled.txt)
Attached.

kawaegle, 2021-08-31:
hello,

Have you try to change the type of the service ? by default I set it as idle (because something waiting to finish but never finish) maybe if you try to change it and set it to **simple** or **dbus**.

The **wantedby** is set by default with **default.target** maybe on your install the default target is not defined (pretty sure that's impossible to not set default but maybe)

hope that's help little I continue to investigate.

thx

moore-bryan, 2021-08-31:
I have not and am not sure how to do that; any chance it's in the install notes?

kawaegle, 2021-09-02:
Hello 

Did you try with my own service ?

https://github.com/OppaiWeeb/asus-touchpad-numpad-driver

moore-bryan, 2021-09-02:
Just did and I have the same results; i.e., blank screen. `dmseg` contains nothing strange and I don't know enough about `journalctl` to use it nor do I know if it would even be useful.

If I _manually_ start the service after logging in, everything functions properly, but if I don't manually disable it before logging out or shutting down, the laptop will be unusable at boot until I switch ttys and disable the service, which will let me reboot and login.

zara86, 2021-09-03:
I'm using Wayland instead of X11, but I have the same problem on Arch.
It's needed a reboot with live iso and chroot to disable the service, in orderr to be able to boot my OS again.
I confirm that if I manually start the service after logging in, all work fine. Strange :\

moore-bryan, 2021-09-03:
Interesting... That would suggest it's not an X issue, but something else...

kawaegle, 2021-09-04:
maybe it's an issue with the python file at start up with systemd (I mean an error in the service file) and so all the systemd init fail because of something 

(but that really strange that append on your computer and not mine (I start and can use my numpad)

```service
[Unit]
Description=Asus Touchpad to Numpad Handler

[Service]
Type=idle
ExecStart=/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py m433ia 40
StandardInput=tty-force
TimeoutSec=5

[Install]
WantedBy=default.target
```

this is my services file. 
Maybe check if my version work ?

moore-bryan, 2021-09-04:
No change here. 

The only difference in the service files between the one you provide above and the one that was installed is that the one that installs via the script contains a `After=sddm.service display-manager.service` line in [Unit] and the number "6" rather than "40" on the ExecStart line.

kawaegle, 2021-09-05:
The difference between 6 and 40 is QWERTY and AZERTY keyboard

I was guessing the after services made the services crash for waiting for a services that doesn't exist

I'll continue to investigate but that a strange issue maybe I'll install Solus os and try by my own

moore-bryan, 2021-09-05:
That explains that, then... Thank you for your persistent help getting this working for us!

moore-bryan, 2021-09-10:
Just to report back... For unrelated reasons I had to switch back to a Fedora install and this works fine there; the corner is a little "off"—that is, my finger has to "slide in" from the right into the _very_ top of the touchpad, not where the symbol is—but that is a **minor** gripe. :-)

moore-bryan, 2021-09-15:
Are you sure this should be closed @mohamed-badaoui; is it resolved/fixed?

mohamed-badaoui, 2021-09-15:
Sorry reading your last comment, I thought it was resolved?
So the problem still exists for Arch and Solus OS.

moore-bryan, 2021-09-15:
Solus, for sure; I do not currently use Arch, so I don't know. I'm no
longer on Solus at the moment, so I just wanted to let you know I have no
way to further debug...
--
*B.Moore*

mohamed-badaoui, 2021-09-16:
I think I have found the reason for this random boot failure on a few systems.

When starting the python script we try to find the correct i2c interface id .. and we only try it 5 times with 0.1 second between each try otherwise we exit with a SIGNAL HANGUP status (sys.exit(1) https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/blob/main/asus_touchpad.py#L64)

The solution therefore consists in increasing the number of tests or the time spent sleeping between each try.

@see https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/65

moore-bryan, 2021-09-16:
Interesting... I hope someone can check and make sure this works for them; @ilSant0?

mohamed-badaoui, 2021-09-19:
For some distro, there is an issue with the service startup. It does not depend on the python script itself.
At this moment, service fails mainly to start on Mint, Pop!Os, ElementaryOs, and SolusOs.

I have just add an error log file to save service error on startup.
Please to uninstall/install, reboot and check the file
```
cat /var/log/asus_touchpad_numpad-driver/error.log
```

related issues: https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/66 https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/65 https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/28

zenusr, 2021-09-19:
I have just reported that two changes are needed in the service file on my Mint 20.2 system for the service to start reliably on system startup or restart. I had 100% failure prior to applying the changes, now 100% success. Please, take a look at the suggested solution in #68.

closed 2023-08-11 by mohamed-badaoui