mohamed-badaoui/asus-touchpad-numpad-driver #62: The backlight is not working Zenbook UM425

anton-voronko, 2021-09-05:
The Numpad is working, but the backlight doesn't. 
I have tried manually enabling it by this command: 
_i2ctransfer -f -y " + device_id + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad_
with correct "device_id" and it doesn't work. 

Also, I have tried @kawaegle script for testing the backlight, and the result is the same with different values.  

I think that enabling backlight in the UM425 has another register address, and I need to send some other I2C data. 
How can I find exactly this I2C sequence?

kawaegle, 2021-09-05:
Hey @anton-voronko

Did you try to check all dependency (check in the readme with your distro)

And did you also try to launch the script manually and so get some output to check if the i2c command fail or not

mohamed-badaoui, 2021-09-10:
Hi

This script was installed and mentioned as working for this model by @haakov. 
Check his PR at https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/pull/21

So the current ic2 sequence should be right.

Can you share more details on this:
- what is your linux Operation System?
- return of command sudo i2cdetect -l
- which device_id did you try?
- the entire command of  i2ctransfer
- the error message displayed

anton-voronko, 2021-09-14:
Hello, @kawaegle and @mohamed-badaoui !  

@kawaegle 
Yes, I have checked all dependecy.
Yes, I tried manually launch the script, and i2c command doesn't show any errors and messages. 

@mohamed-badaoui 
Have checked @haakov #21. I don't have any errors when launching the script. Also, I changed the resolution as was described in this post, and still, the backlight is not working. And it would be strange if by changing this code (x > 0.95 * maxx) and (y < 0.05 * maxy), the backlight start working. 

I am using:  Ubuntu 20.04.3 LTS (Focal Fossa) 5.11.0-34-generic

[image]
[image]

Also, I have played with i2c-tools. And I see that when I am dumping address 0x15 I have all 0xff values. I  have tried to write some other data, and they were successfully written. So I expect that the i2ctransfer command is working.

mohamed-badaoui, 2021-09-15:
If you see the last reported issue, it seems to work fine when the script is manually run on UM425
https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/65

Maybe you should try comparing your list of i2c interfaces with his and see if there is a particular difference.

mohamed-badaoui, 2021-09-20:
For this model, it was tested as working.
The i2c command is used to turn on the touchpad backlights and it is a prerequisite.
If i2ctranfer command does not work manually, so it is a system issue.

closed 2021-09-20 by mohamed-badaoui