mohamed-badaoui/asus-touchpad-numpad-driver #63: Traceback y not defined

martininsulander, 2021-09-09:
Hi,
I don't have the traceback available. But in the code I found something that looked wrong.

In /usr/bin/asus_touchpad_numpad.py there looks to me (and my traceback at computer startup), that x and y could be undefined:
```
      # During tap #
      if finger == 1:
          finger = 2
          try:
              # first row
              if y < 0.25 * maxy:
```
if not these have been matched:
```
        # Get x position #
        if e.matches(EV_ABS.ABS_MT_POSITION_X):
            x = e.value
            continue
        # Get y position #
        if e.matches(EV_ABS.ABS_MT_POSITION_Y):
            y = e.value
            continue
```
Default values for x and y maybe could help, or an indicator if this has happened.

Also while at it, the try-catch-case this is in looks like it is only used for the last rows, so try could be moved there instead.

mohamed-badaoui, 2021-09-15:
Hi,

Events ABS_MT_POSITION_X and ABS_MT_POSITION_Y always precede BTN_TOOL_FINGER events So x and y are always defined before checking their value to determinate current touch coordinate.

closed 2021-09-15 by mohamed-badaoui

martininsulander, 2021-09-15:
The issue is that it has happened me at least once, and then the numpad was lit until I restarted the application.

And, I would consider it bad programming practice to blindly trust complex drivers like this.

mohamed-badaoui, 2021-09-15:
Hi,

It is a collaborative project. Do not hesitate to submit any PR that would improve the readability and maintainability of this project
We're not all Python developers, so we're doing our best to find a workaround to make the Asus touchpad work.

Thank you for your indulgence

martininsulander, 2021-09-20:
Hi Mohamed, I did a pull-request with some cleanup of the code if you care to check :)

mohamed-badaoui, 2021-09-20:
Thank you!
I need some time to test it and review it :)
I will soon, I'm a little busy right now