mohamed-badaoui/asus-touchpad-numpad-driver #65: Load at start up and a few suggestions

IGrayfoxI, 2021-09-12:
I just installed it on my new UM425UAZ laptop and it mostly works with the only issues being Fn+F8 not enabling the numpad.
The top right on the touchpad enables the numpad no issues.

Is their anyway to make the driver load up at start up?
I can start the service it manually with "sudo service asus_touchpad_numpad start"
But when starting it again the brightness of the touchpad goes back to lv1 instead of where I left it(in my case Lv3)

While the Fn+F8 issue does not bother me,  it might be an issue for someone else.
Have confirmed the issue mentioned in #52 

Is it possible to rebind the "Calculator" key when the numpad is disabled.

When using the numpad in windows, you require a 1second hold of the numpad button to turn the numpad on, this would be nice to have here as well to prevent accidental activation

Thank you.

mohamed-badaoui, 2021-09-15:
Hi,

The install script adds a systemd service that automatically runs the touchpad python script when the machine starts up.

You don't need to run the script manually.
If it is not started on the boot machine, you probably have a problem with the service.

What is your exploitation system ?
By default on the service configuration file, the timeout is set to 5 seconds. It might be too small.
You can try increasing it to 10 or 20 seconds and then run uninstall.sh and install.sh

IGrayfoxI, 2021-09-15:
Thanks for the response.

Think their might be a conflict between it something.
I disabled all start up items.
It started to work on reboot, I then rebooted a it stopped working.

Any way to bring up logs to troubleshoot?

mohamed-badaoui, 2021-09-15:
ok so you re might experiencing services conflict.

Try to defer the service execution as late as possible, after graphic display manager for example.
And add "After" property line in service file, in the unit section like this:

```
[Unit]
Description=Asus Touchpad to Numpad Handler
After=gdm.service
```

I defer touchpad launch after gdm service start (one of the last started service)

to identify your graphic manager service name:
```
systemctl list-units --type=service | grep -i displa
```

IGrayfoxI, 2021-09-15:
Uninstalled
added the After=lightdm.service
Installed with the new line added to the service file.
Worked on two reboots, did not work on the third reboot.
sudo systemctl status asus_touchpad_numpad.service when the touchpad works on boot up
[image]

sudo systemctl status asus_touchpad_numpad.service when the touchpad fails to work on boot up
[image]

mohamed-badaoui, 2021-09-15:
Could you add in the service section **SuccessExitStatus=SIGHUP**

```
[Service]
Type=simple
ExecStart=/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py $LAYOUT $PERCENTAGE_KEY
StandardInput=tty-force
TimeoutSec=10
Restart=on-failure
SuccessExitStatus=SIGHUP
```

IGrayfoxI, 2021-09-16:
[image]`

This is the code in the service file
```
[Unit]
Description=Asus Touchpad to Numpad Handler
After=lightdm.service

[Service]
Type=simple
ExecStart=/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py m433ia 6
StandardInput=tty-force
TimeoutSec=10
Restart=on-failure
SuccessExitStatus=SIGHUP

[Install]
WantedBy=default.target
```

mohamed-badaoui, 2021-09-16:
ok great. So with SuccessExitStatus=SIGHUP parameter we could bypass this  killed process unexplained case?
Could we close this issue?

mohamed-badaoui, 2021-09-16:
fixed.
When the service failed to start at machine boot, just adjust the values of your config file in numpad_layouts directory. 

```
# Number of tries to identify the interface number
try_times = 5
try_sleep = 0.1
```

IGrayfoxI, 2021-09-18:
Tried adding those two lines to the m433ia script for my touchpad, my OS is Linux Mint 20.2.

zenusr, 2021-09-19:
I have just reported that two changes are needed in the service file on my Mint 20.2 system for the service to start reliably on system startup or restart. I had 100% failure prior to applying the changes, now 100% success. Please, take a look at the suggested solution in #68.

IGrayfoxI, 2021-09-20:
Seems to be working so far as per the details of #68

closed 2021-09-20 by IGrayfoxI

LordEvron, 2022-01-26:
@IGrayfoxI  I have the same issue.. what is the solution? it seems issue #68 has been deleted!