mohamed-badaoui/asus-touchpad-numpad-driver #66: driver fails to load on restart

chowdri, 2021-09-17:
I use UX581GV laptop model and the driver works when installed. However, the driver does not work after restart, it has to be installed again.

mohamed-badaoui, 2021-09-17:
Hi,

I presume the configuration file you use is ux581l.
Open its configuration file https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/blob/main/numpad_layouts/ux581l.py

And adjust the value :
```
# Number of tries to identify the interface number
try_times = 5
try_sleep = 0.1
``` 

Set try_sleep to 0.5 and try_times to 10.

That way on the reboot the service will try 10 times to find your right i2c interface every 0.5 second.
On some system, the service require more time to identify the i2c interface.

chowdri, 2021-09-17:
the instruction gives the following output, and the numpad isn't working at the time of this output

● asus_touchpad_numpad.service - Asus Touchpad to Numpad Handler
     Loaded: loaded (/lib/systemd/system/asus_touchpad_numpad.service; enabled; vendor preset>
     Active: inactive (dead) since Fri 2021-09-17 22:58:09 IST; 8min ago
    Process: 1032 ExecStart=/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py ux581l 6 >
   Main PID: 1032 (code=killed, signal=HUP)

Sep 17 22:58:09 ChowdriLaptop systemd[1]: Started Asus Touchpad to Numpad Handler.
Sep 17 22:58:09 ChowdriLaptop systemd[1]: asus_touchpad_numpad.service: Succeeded.

chowdri, 2021-09-17:
I reinstalled using the updated repository, the brightness of the numpad appears dimmer. Changing the try_times and try_sleep parameters did not change the outcome.

mohamed-badaoui, 2021-09-17:
Can you increase the TimeoutSec of this file tp 10 for example.
```
sudo nano /etc/systemd/system/asus_touchpad_numpad.service
```
```
[Service]
Type=simple
ExecStart=/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py (depends on your configuration)
StandardInput=tty-force
**TimeoutSec=10**
```

Be sure to run uninstall.sh then install.sh

chowdri, 2021-09-18:
It's already 10.
[Unit]
Description=Asus Touchpad to Numpad Handler

[Service]
Type=simple
ExecStart=/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py ux581l 6
StandardInput=tty-force
TimeoutSec=10

[Install]
WantedBy=default.target

chowdri, 2021-09-18:
I'm sorry to bother you with this, just getting familiar with open source. It'll take sometime to get conversant with all the frameworks and languages.

mohamed-badaoui, 2021-09-18:
No problem, we will solve your issue.

What does these commands return:

```
sudo i2cdetect -l
```
and this one:
```
grep -B1 -A5 -i -E "(elan|asu)" /proc/bus/input/devices
```

chowdri, 2021-09-18:
sudo i2cdetect -l 

i2c-3	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-10	i2c       	AUX A/DDI A/PHY A               	I2C adapter
i2c-1	smbus     	SMBus I801 adapter at efa0      	SMBus adapter
i2c-8	i2c       	i915 gmbus misc                 	I2C adapter
i2c-6	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-13	i2c       	NVIDIA i2c adapter 1 at 1:00.0  	I2C adapter
i2c-4	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-11	i2c       	AUX B/DDI B/PHY B               	I2C adapter
i2c-2	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-0	i2c       	NVIDIA GPU I2C adapter          	I2C adapter
i2c-9	i2c       	i915 gmbus dpd                  	I2C adapter
i2c-7	i2c       	i915 gmbus dpc                  	I2C adapter
i2c-14	i2c       	NVIDIA i2c adapter 7 at 1:00.0  	I2C adapter
i2c-5	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-12	i2c       	AUX D/DDI D/PHY D               	I2C adapter

grep -B1 -A5 -i -E "(elan|asu)" /proc/bus/input/devices

I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input27
U: Uniq=
H: Handlers=rfkill kbd event4 
B: PROP=0
B: EV=100013
B: KEY=1000000080000 0 0 0 0 181606f00900000 8780027801701000 e000000000000 0
--
I: Bus=0018 Vendor=04f3 Product=29b6 Version=0100
N: Name="ELAN9008:00 04F3:29B6"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-4/i2c-ELAN9008:00/0018:04F3:29B6.0001/input/input37
U: Uniq=
H: Handlers=mouse0 event9 
B: PROP=2
B: EV=1b
B: KEY=400 0 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=29b6 Version=0100
N: Name="ELAN9008:00 04F3:29B6 UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-4/i2c-ELAN9008:00/0018:04F3:29B6.0001/input/input38
U: Uniq=
H: Handlers=event10 
B: PROP=0
B: EV=9
B: ABS=10000000000
--
I: Bus=0018 Vendor=04f3 Product=29b6 Version=0100
N: Name="ELAN9008:00 04F3:29B6 UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-4/i2c-ELAN9008:00/0018:04F3:29B6.0001/input/input39
U: Uniq=
H: Handlers=event28 
B: PROP=0
B: EV=100001

I: Bus=0018 Vendor=04f3 Product=29b6 Version=0100
N: Name="ELAN9008:00 04F3:29B6 Stylus"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-4/i2c-ELAN9008:00/0018:04F3:29B6.0001/input/input40
U: Uniq=
H: Handlers=mouse1 event29 
B: PROP=2
B: EV=1b
B: KEY=c03 0 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=29b6 Version=0100
[... 114 lines cut ...]
B: PROP=5
B: EV=1b
B: KEY=e420 10000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=3101 Version=0100
N: Name="ELAN1406:00 04F3:3101 Keyboard"
P: Phys=i2c-ELAN1406:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-3/i2c-ELAN1406:00/0018:04F3:3101.0004/input/input57
U: Uniq=
H: Handlers=sysrq kbd event20 leds 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff800000000007ff febeffdfffefffff fffffffffffffffe

uname -r 

Linux ChowdriLaptop 5.11.0-34-generic #36~20.04.1-Ubuntu SMP Fri Aug 27 08:06:32 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

lsb_release -a

No LSB modules are available.
Distributor ID:	Elementary
Description:	elementary OS 6 Odin
Release:	6
Codename:	odin

mohamed-badaoui, 2021-09-18:
ok allright.

Can you share the result of this command (it search for a keyboard device)

```
grep -B1 -A5 -i -E "(elan|asu|keyboard)" /proc/bus/input/devices
```

chowdri, 2021-09-18:
grep -B1 -A5 -i -E "(elan|asu|keyboard)" /proc/bus/input/devices

I: Bus=0011 Vendor=0001 Product=0001 Version=ab83
N: Name="AT Translated Set 2 keyboard"
P: Phys=isa0060/serio0/input0
S: Sysfs=/devices/platform/i8042/serio0/input/input3
U: Uniq=
H: Handlers=sysrq kbd event3 leds 
B: PROP=0
--
I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input27
U: Uniq=
H: Handlers=rfkill kbd event4 
B: PROP=0
B: EV=100013
B: KEY=1000000080000 0 0 0 0 181606f00900000 8780027801701000 e000000000000 0
--
I: Bus=0018 Vendor=04f3 Product=29b6 Version=0100
N: Name="ELAN9008:00 04F3:29B6"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-4/i2c-ELAN9008:00/0018:04F3:29B6.0001/input/input37
U: Uniq=
H: Handlers=mouse0 event9 
B: PROP=2
B: EV=1b
B: KEY=400 0 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=29b6 Version=0100
N: Name="ELAN9008:00 04F3:29B6 UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-4/i2c-ELAN9008:00/0018:04F3:29B6.0001/input/input38
U: Uniq=
H: Handlers=event10 
B: PROP=0
B: EV=9
B: ABS=10000000000
--
I: Bus=0018 Vendor=04f3 Product=29b6 Version=0100
N: Name="ELAN9008:00 04F3:29B6 UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-4/i2c-ELAN9008:00/0018:04F3:29B6.0001/input/input39
U: Uniq=
H: Handlers=event28 
B: PROP=0
B: EV=100001

I: Bus=0018 Vendor=04f3 Product=29b6 Version=0100
N: Name="ELAN9008:00 04F3:29B6 Stylus"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-4/i2c-ELAN9008:00/0018:04F3:29B6.0001/input/input40
U: Uniq=
H: Handlers=mouse1 event29 
B: PROP=2
B: EV=1b
B: KEY=c03 0 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=29b6 Version=0100
N: Name="ELAN9008:00 04F3:29B6 UNKNOWN"
P: Phys=i2c-ELAN9008:00
S: Sysfs=/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-4/i2c-ELAN9008:00/0018:04F3:29B6.0001/input/input42
U: Uniq=
H: Handlers=event30 
B: PROP=0
B: EV=9
B: ABS=10000000000
--
I: Bus=0018 Vendor=04f3 Product=29b6 Version=0100
[... 108 lines cut ...]
I: Bus=0018 Vendor=04f3 Product=3101 Version=0100
N: Name="ELAN1406:00 04F3:3101 Keyboard"
P: Phys=i2c-ELAN1406:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-3/i2c-ELAN1406:00/0018:04F3:3101.0004/input/input57
U: Uniq=
H: Handlers=sysrq kbd event20 leds 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff800000000007ff febeffdfffefffff fffffffffffffffe
--
I: Bus=0005 Vendor=046d Product=b019 Version=0003
N: Name="MX Master 2S Keyboard"
P: Phys=24:41:8c:12:f0:92
S: Sysfs=/devices/virtual/misc/uhid/0005:046D:B019.0006/input/input65
U: Uniq=d7:02:84:c6:98:ea
H: Handlers=sysrq kbd event8 
B: PROP=0
--
I: Bus=0005 Vendor=046d Product=b34f Version=0021
N: Name="G613 Keyboard"
P: Phys=24:41:8c:12:f0:92
S: Sysfs=/devices/virtual/misc/uhid/0005:046D:B34F.0008/input/input72
U: Uniq=d9:db:6c:f6:8b:7e
H: Handlers=sysrq kbd event262 leds 
B: PROP=0

mohamed-badaoui, 2021-09-18:
Ok your input devices list are right.

It(s an issue related to the OS.
There is an other opened issue with the same problem (with Solus OS)

I invite you to follow
https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/61

mohamed-badaoui, 2021-09-19:
For some distro, there is an issue with the service startup. It does not depend on the python script itself.
At this moment, service fails mainly to start on Mint, Pop!Os, ElementaryOs, and SolusOs.

I have just add an error log file to save service error on startup.
Please to uninstall/install, reboot and check the file
```
cat /var/log/asus_touchpad_numpad-driver/error.log
```

related issues: https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/66 https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/28 https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/61

zenusr, 2021-09-19:
I have just reported that two changes are needed in the service file on my Mint 20.2 system for the service to start reliably on system startup or restart. I had 100% failure prior to applying the changes, now 100% success. Please, take a look at the suggested solution in #68.