mohamed-badaoui/asus-touchpad-numpad-driver #68: [UM431DA / Mint 20.2] Works perfectly, startup failure fixed by changing the service file

zenusr, 2021-09-19:
Great job, Mohamed, thanks a lot for this elegant solution.

1 - Please add UM431DA to the list of supported models, it works perfectly with the ux433fa.py layout file

2 - On Mint 20.2, I have also had challenges trying to make it work on system startup/restart. The service would start manually without issues, but would not start automatically on system start no matter what. After combing through all the related information here, I have  managed to come up with a solution candidate.

Yes, it is definitely a timing issue, but the latest suggested solution goes down the wrong path. After testing many permutations of modifications, I did not need to modify the layout file at all in the end, only the service file.

The contents of my service file:

`
[Unit]
Description=Asus Touchpad to Numpad Handler

[Service]
Type=simple
ExecStartPre=/bin/sleep 2
ExecStart=/usr/share/asus_touchpad_numpad-driver/asus_touchpad.py $LAYOUT $PERCENTAGE_KEY
StandardInput=tty-force
StandardOutput=/var/log/asus_touchpad_numpad-driver/error.log
StandardError=/var/log/asus_touchpad_numpad-driver/error.log
TimeoutSec=5
Restart=on-failure

[Install]
WantedBy=default.target

`

Both the "ExecStartPre" and "Restart" lines are needed for the fix, missing either results in failure. It works 100% of the time with ExecStartPre  >= 2 , "1" is not enough for me!

I hope this works for everyone.

Refs: https://github.com/haakov/asus-touchpad-numpad-driver/pull/1, #18, #21, #61, #65, #66

Service file:
[asus_touchpad.service.zip](https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/files/7192026/asus_touchpad.service.zip)

mohamed-badaoui, 2021-09-19:
Good news!
It seems that generally for the majority of system configurations, we don't need to add any sleep time (ExecStartPre=/bin/sleep 2)
So I think that it s could be a nice workaround for those with startup failure.

Thank you :+1:

kawaegle, 2021-09-20:
I want to say thank you  @zenusr for the status files, I have no experience and my solution was horrible.

so thank you to help us with that file.

mohamed-badaoui, 2021-09-20:
readme updated

closed 2021-09-20 by mohamed-badaoui