mohamed-badaoui/asus-touchpad-numpad-driver #71: Working but no light up solution (maybe)

Tambup, 2021-10-01:
Hi,

first thing is that the driver is working on my laptop, that is a "X430FA", with the "ux433fa" layout.
The only problem is that i can't see symbols on the touchpad. I solved changing the activation function in this way:
```python3
def activate_numlock(brightness):
    numpad_cmd = "i2ctransfer -f -y " + device_id + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 " + BRIGHT_VAL[brightness] + " 0xad"
    events = [
        InputEvent(EV_KEY.KEY_NUMLOCK, 1),
        InputEvent(EV_SYN.SYN_REPORT, 0)
    ]
    udev.send_events(events)
    d_t.grab()
    subprocess.call(numpad_cmd, shell=True)
    change_brightness(1)  # !!! THAT ONE IS THE NEW LINE !!!
```

This way it is fully working. My question is: why doesn't it already work this way? Incompatibility with other devices?
However, if this fix can be accepted, I can fork and add this fix and the update to the README with the new device.

As this is very useful, if it's ok for you, I'd like to add this to the AUR (I will always use the last commit until you make a release) to make it immediate to install on arch. In this case, in my opinion, can be recommended to add a license.

wkynrocks, 2021-10-07:
I had the same problem and it worked for me as well. Please remind the method name is change_bright (not change_brightness though).

mohamed-badaoui, 2021-11-17:
Hi,
Keep in mind that is very difficult to help because i dont own all the asus models and I do not have tested on each linux distribution. 
I just know that this issue was never reported before. 
So is it a real issue, an issue for a particular model or specific linux OS.. hard to say ☹️

Tambup, 2021-11-17:
@mohamed-badaoui makes sense. Let's wait if someone else has this problem to check.

glebtv, 2022-02-27:
This didn't solve it for me. But my backlight started working with this driver after a reboot from windows (keys were working).

ASUS Zenbook 14X UX5401EA using m433ia qwerty layout, arch linux

TheMY3, 2022-04-08:
Didn't solve to me :disappointed:

Hilicot, 2022-06-02:
> @mohamed-badaoui makes sense. Let's wait if someone else has this problem to check.

I had this problem on an Asus Vivobook M3401QC with Ubuntu 22 LTS. This code fixes my problem

charlesleaub, 2023-03-02:
It worked perfectly for me it should be merged with this line added!

natoch, 2023-09-05:
Thank you! Worked well for me on the UX3402VA model.

iuyfu, 2023-12-10:
> Hi,
> 
> first thing is that the driver is working on my laptop, that is a "X430FA", with the "ux433fa" layout. The only problem is that i can't see symbols on the touchpad. I solved changing the activation function in this way:
> 
> ```python
> def activate_numlock(brightness):
>     numpad_cmd = "i2ctransfer -f -y " + device_id + " w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 " + BRIGHT_VAL[brightness] + " 0xad"
>     events = [
>         InputEvent(EV_KEY.KEY_NUMLOCK, 1),
>         InputEvent(EV_SYN.SYN_REPORT, 0)
>     ]
>     udev.send_events(events)
>     d_t.grab()
>     subprocess.call(numpad_cmd, shell=True)
>     change_brightness(1)  # !!! THIS ONE IS THE NEW LINE !!!
> ```
> 
> This way it is fully working. My question is: why doesn't it already work this way? Incompatibility with other devices? However, if this fix can be accepted, I can fork and add this fix and the update to the README with the new device.
> 
> As this is very useful, if you agree, I'd like to add this to the AUR (I will always use the last commit until you make a release) to make it immediate to install on arch. In this case, in my opinion, can be recommended to add a license.

Hi, 

Is this solution easy to apply for regular users ? I have no particular knowledge nor skills but I am facing this particularly annoying issue and it seems, you are the only solution on the internet !

Tambup, 2023-12-11:
> Hi,
>
> Is this solution easy to apply for regular users ? I have no particular knowledge nor skills but I am facing this particularly annoying issue and it seems, you are the only solution on the [...]

@iuyfu yes, it is. Just go to [this function](https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/blob/bfbd282025e1aeb2c805a881e01089fe55442e7f/asus_touchpad.py#L152) (on your local laptop) and add the following line at the end of the function:
```python
change_brightness(1)
```

iuyfu, 2023-12-12:
> 
> 
> @iuyfu yes, it is. Just go to [this function](https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/blob/bfbd282025e1aeb2c805a881e01089fe55442e7f/asus_touchpad.py#L152) (on your local laptop) and add the following line at the end of the function:
> 
> ```python
> change_brightness(1)
> ```

Hey, thanks for answering. I was not clear enough haha, I have no clue how to even access this function, could you tell me ?

Tambup, 2023-12-12:
Hi!
What do you mean by "access"?

iuyfu, 2023-12-13:
Hi,

I think I cannot express myself clearly because I do not really know what I am talking about, let me try to be understood.

You published a few lines of code that will fix this issue. For that to work, I need to copy/paste this command somewhere, don’t I?
This is the part where I’m stuck. I do not know where I am supposed to copy/paste this command. Do I have to use the command prompt?

Tambup, 2023-12-13:
Hi @iuyfu, maybe now I got your question, let's see:
No, you don't have to type in the command prompt, but to change the code of that function in the source code.
All you need to do, once the package is installed, is to change the file [asus_touchpad.py](https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/blob/main/asus_touchpad.py) in your laptop, by adding that specific line in the file itself. Rephrased, you need to edit that file in the same way you change a text document.

ransur0t, 2023-12-15:
Hi.  Foremost, thank you for the driver script!  I've installed it on model UM3402ya with m433ia layout and the numpad activates but the brightness function does not work -- numpad backlight is not visible despite making the edit to asus_touchpad.py ...

Is there an additional edit that needs to be made?

Tambup, 2023-12-16:
Hi @ransur0t.
It's difficult to backtrack, but my first idea for your original problem is that probably you have to restart the service after changing it.

About the calculator launch, unfortunately I can't guess anything.

ransur0t, 2023-12-16:
@Tambup thanks -- yet, I was sure to reboot, multiple times!  When I used the debug command it suddenly worked ... which is essentially manually triggering the driver script.

Regarding the calculator launch, I think that the other repo installation script made a change to the launcher association between the calc and the touchpad ... I'll have to dig into that.

Unfortunately I have a larger issue -- the MediaTek 7922 802.11ax WiFi adapter has pronounced issues with reliable speeds, sometimes slowing to a trickle, and other times not activating after wake.

It is a known issue, so either I keep the laptop and replace it with an Intel AX210 or do a return and continue my search for a replacement for my aging XPS 13 (9360)!

vazhnov, 2024-03-15:
> As this is very useful, if you agree, I'd like to add this to the AUR (I will always use the last commit until you make a release) to make it immediate to install on arch. In this case, in my opinion, can be recommended to add a license.

@Tambup , I've created dedicated issue about license: https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/119

iuyfu, 2024-04-11:
> Hi!
> What do you mean by "access"?

Hi, took me a long time to answer haha.

I am not familiar at all with github and I couldn't understand how to "use" this solution.
I clicked on your link and downloaded the .py file but I have no clue what to do with it.

I might have missed an explanation page or something?

ndelizoreux, 2024-05-24:
Hi everyone,
I have an asus zenbook 14 UX3402ZA , and I have the layout m433ia, and my linux distro is Arch.
I had the same issue with the backlight, and the new line of code did the job and now everything works fine.

ndelizoreux, 2024-05-24:
> Hi, took me a long time to answer haha.
>
> I am not familiar at all with github and I couldn't understand how to "use" this solution. I clicked on your link and downloaded the .py file but I have no [...]

Hello @iuyfu 
Before to install, go into the `asus_touchpad.py` and add under line 166 `change_brightness(1)`, don't forget to respect the indent. Then reinstall the driver. 
Have a good day !

ignSKRRRTT, 2024-06-20:
> Hello @iuyfu Before to install, go into the `asus_touchpad.py` and add under line 166 `change_brightness(1)`, don't forget to respect the indent. Then reinstall the driver. Have a good day !

This works, but when you switch out of numpad mode the numpad is still lit up. Any way to fix?