mohamed-badaoui/asus-touchpad-numpad-driver #76: ROG Zephyrus GX501 support

hvico, 2021-11-14:
Hi,

Are there any plans to support the ASUS ROG Zephyrus GX501?

It has one of this numpads. I offer mine to do testing or checks if it is required. Thanks in advance!

[image]

mohamed-badaoui, 2021-11-17:
Hi,

Did you try to install it selectin the choice 1

Select models keypad layout:
1) gx701.py
2) m433ia.py
3) ux433fa.py
4) ux581l.py
5) Quit
Please enter your choice

GX501 model should be compatible with GX701 configuration.

hvico, 2021-11-17:
Hi! Thanks for your response.

This is what I get as soon as I tap the top right corner of the trackpad (selected model gx701):

INFO:Pad:Detect keyboard from N: Name="AT Translated Set 2 keyboard"
INFO:Pad:Set keyboard 4 from H: Handlers=sysrq kbd event4 leds
INFO:Pad:Detect touchpad from N: Name="ELAN1404:00 04F3:30A7 Touchpad"
INFO:Pad:Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-1/i2c-ELAN1404:00/0018:04F3:30A7.0006/input/input19
INFO:Pad:Set touchpad id 19 from H: Handlers=mouse1 event19
DEBUG:Pad:Touchpad min-max: x 0-1764, y 0-2265
DEBUG:Pad:finger down at x 1708 y 4
Traceback (most recent call last):
  File "/home/hvico/.local/lib/python3.6/site-packages/libevdev/device.py", line 809, in grab
    self._libevdev.grab()
  File "/home/hvico/.local/lib/python3.6/site-packages/libevdev/_clib.py", line 556, in grab
    raise OSError(-r, os.strerror(-r))
OSError: [Errno 16] Device or resource busy

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./asus_touchpad.py", line 255, in <module>
    activate_numlock(brightness)
  File "./asus_touchpad.py", line 159, in activate_numlock
    d_t.grab()
  File "/home/hvico/.local/lib/python3.6/site-packages/libevdev/device.py", line 811, in grab
    raise DeviceGrabError()
libevdev.device.DeviceGrabError

mohamed-badaoui, 2021-11-17:
Device or resource budy?

Does the script are currently running in background.
Please be sure to uninstall it before running the script manually.

Run ./uninstall.sh before
Then try
 
```
LOG=DEBUG sudo -E ./asus_touchpad.py gx701 6
```

hvico, 2021-11-17:
Hi! Thanks for the advice. It seems I've had a previous version installed as you suggested. After doing an uninstall.sh an fresh install now I can activate de numpad tapping the top right corner, but that doesn't turn the backlight on with the numbers and symbols. Is that supported?

Thanks again,

mohamed-badaoui, 2021-11-17:
Sorry I dont't know. 
I only know that It is supported for the gx701model.
it would be weird if the i2c hexadecimal command is different for your model.

hvico, 2021-11-27:
UPDATE: I was able to fix it for my Gx501 by changing this line in the code:

BRIGHT_VAL = [hex(val) for val in [1]]

Previously it had 3 different values. Running tests/test_brightness.py I realized in my model the only value that did something was 0 (turn the backlight off) and 1 to turn it on. The intermediate values to set brightness settings doesn't seem to apply in my case.

So now it works.

Thanks,