mohamed-badaoui/asus-touchpad-numpad-driver #81: UX431 Numpad Mod - No Backlight

ZdenoF, 2021-12-02:
Hi guys,

I think this one is suitable as a new bug for this thread...

I'm pretty new to linux (finally decided to get rid of windows after many years). I'm runnig on Ubuntu 21.10 now...

I have Asus Zenbook 14 UX431DA which originally came with standard touchpad with fingerprint sensor. So I decided to buy numpad mod hardware piece for my notebook. Succesfully installed yesterday, also did install driver package. Everything is working correctly but without backlight.
When I press the numlock button on it, I can type the numbers and symbols, but I'm not able to toggle the brightness (backlight), so I don't see anything on the touchpad.
As I am working a lot in excel I really need to get a numpad working properly. So I hope that somebody could help me with this.
Thanks, Zee

ZdenoF, 2021-12-02:
I have also run this command : grep -B1 -A5 -i -E "(elan|asu)" /proc/bus/input/devices

Here the results :
I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input13
U: Uniq=
H: Handlers=rfkill kbd event5 
B: PROP=0
B: EV=100013
B: KEY=1000000080000 0 0 0 0 181606f00900000 8280027801701000 e000000000000 0
--
I: Bus=0018 Vendor=04f3 Product=30f4 Version=0100
N: Name="ELAN1401:00 04F3:30F4 Mouse"
P: Phys=i2c-ELAN1401:00
S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ELAN1401:00/0018:04F3:30F4.0001/input/input16
U: Uniq=
H: Handlers=mouse0 event8 
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=30f4 Version=0100
N: Name="ELAN1401:00 04F3:30F4 Touchpad"
P: Phys=i2c-ELAN1401:00
S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ELAN1401:00/0018:04F3:30F4.0001/input/input17
U: Uniq=
H: Handlers=mouse1 event9 
B: PROP=5
B: EV=1b
B: KEY=e520 10000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=30f4 Version=0100
N: Name="ELAN1401:00 04F3:30F4 Keyboard"
P: Phys=i2c-ELAN1401:00
S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ELAN1401:00/0018:04F3:30F4.0001/input/input18
U: Uniq=
H: Handlers=sysrq kbd event10 leds 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff800000000007ff febeffdfffefffff fffffffffffffffe
--
I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="Asus Touchpad/Numpad"
P: Phys=
S: Sysfs=/devices/virtual/input/input25
U: Uniq=
H: Handlers=kbd event14 
B: PROP=0

Tried to switch backlight manualy, but I wasn't succesfull :(
sudo i2ctransfer -f -y 0 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x01 0xad

StarterX4, 2022-03-30:
Same here with ROG Strix G15 2021 G513QY:
````
[starterx4@doadgrz ~]$ grep -B1 -A5 -i -E "(elan|asu)" /proc/bus/input/devices
I: Bus=0003 Vendor=0b05 Product=1866 Version=0110
N: Name="Asus Keyboard"
P: Phys=usb-0000:07:00.3-3/input0
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:07:00.3/usb1/1-3/1-3:1.0/0003:0B05:1866.0002/input/input6
U: Uniq=
H: Handlers=sysrq kbd leds event5 rfkill 
B: PROP=0
--
I: Bus=0019 Vendor=1043 Product=0000 Version=0000
N: Name="Asus Wireless Radio Control"
P: Phys=asus-wireless/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/ATK4002:00/input/input10
U: Uniq=
H: Handlers=kbd event9 rfkill 
B: PROP=0
B: EV=3
--
I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input11
U: Uniq=
H: Handlers=kbd event10 rfkill 
B: PROP=0
B: EV=100013
B: KEY=1000000080000 0 0 0 0 181606f00900000 8280027801701000 e000000000000 0
--
I: Bus=0018 Vendor=04f3 Product=319b Version=0100
N: Name="ASUE120A:00 04F3:319B Mouse"
P: Phys=i2c-ASUE120A:00
S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ASUE120A:00/0018:04F3:319B.0005/input/input14
U: Uniq=
H: Handlers=event11 mouse1 
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=319b Version=0100
N: Name="ASUE120A:00 04F3:319B Touchpad"
P: Phys=i2c-ASUE120A:00
S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ASUE120A:00/0018:04F3:319B.0005/input/input15
U: Uniq=
H: Handlers=event12 mouse2 
B: PROP=5
B: EV=1b
B: KEY=e520 10000 0 0 0 0
--
I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="Asus Touchpad/Numpad"
P: Phys=
S: Sysfs=/devices/virtual/input/input20
U: Uniq=
H: Handlers=kbd event17 
B: PROP=0
````

TheMY3, 2022-04-08:
Same for me, working without backlight
```
I: Bus=0018 Vendor=04f3 Product=30f1 Version=0100
N: Name="ELAN1206:00 04F3:30F1 Mouse"
P: Phys=i2c-ELAN1206:00
S: Sysfs=/devices/platform/AMDI0010:03/i2c-0/i2c-ELAN1206:00/0018:04F3:30F1.0001/input/input7
U: Uniq=
H: Handlers=event5 mouse0 
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=30f1 Version=0100
N: Name="ELAN1206:00 04F3:30F1 Touchpad"
P: Phys=i2c-ELAN1206:00
S: Sysfs=/devices/platform/AMDI0010:03/i2c-0/i2c-ELAN1206:00/0018:04F3:30F1.0001/input/input8
U: Uniq=
H: Handlers=event6 mouse1 
B: PROP=5
B: EV=1b
B: KEY=e520 10000 0 0 0 0
--
I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input12
U: Uniq=
H: Handlers=kbd event10 rfkill 
B: PROP=0
B: EV=100013
B: KEY=1000000080000 0 0 0 0 181606f00900000 8280027801701000 e000000000000 0
--
I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="Asus Touchpad/Numpad"
P: Phys=
S: Sysfs=/devices/virtual/input/input17
U: Uniq=
H: Handlers=kbd event15 
B: PROP=0
```

caterpillarpants, 2022-10-21:
Thanks for working on this!
I have the same issue for my G512L. seems to work, but no backlight.

I tried all the layouts.
Also edited asus_touchpad_numpad_dev and messed with the backlight setting and the default backlight level, but to no avail.

grep -B1 -A5 -i -E "(elan|asu)" /proc/bus/input/devices
`I: Bus=0003 Vendor=0b05 Product=1866 Version=0110
N: Name="Asus Keyboard"
P: Phys=usb-0000:00:14.0-8/input0
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:0B05:1866.0004/input/input19
U: Uniq=
H: Handlers=sysrq kbd event11 leds 
B: PROP=0
--
I: Bus=0003 Vendor=0b05 Product=1866 Version=0110
N: Name="Asus Keyboard"
P: Phys=usb-0000:00:14.0-8/input1
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:0B05:1866.0005/input/input20
U: Uniq=
H: Handlers=sysrq kbd event12 leds 
B: PROP=0
--
I: Bus=0003 Vendor=0b05 Product=1866 Version=0110
N: Name="Asus Keyboard"
P: Phys=usb-0000:00:14.0-8/input2
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:0B05:1866.0006/input/input21
U: Uniq=
H: Handlers=rfkill kbd event13 
B: PROP=0
--
I: Bus=0019 Vendor=1043 Product=0000 Version=0000
N: Name="Asus Wireless Radio Control"
P: Phys=asus-wireless/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/ATK4002:00/input/input22
U: Uniq=
H: Handlers=rfkill kbd event14 
B: PROP=0
B: EV=3
--
I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input23
U: Uniq=
H: Handlers=rfkill kbd event7 
B: PROP=0
B: EV=100013
B: KEY=1000000080000 0 0 0 0 181606f00900000 8280027801701000 e000000000000 0
--
I: Bus=0018 Vendor=04f3 Product=307a Version=0100
N: Name="ELAN1203:00 04F3:307A Mouse"
P: Phys=i2c-ELAN1203:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-3/i2c-ELAN1203:00/0018:04F3:307A.0001/input/input24
U: Uniq=
H: Handlers=mouse0 event8 
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=307a Version=0100
N: Name="ELAN1203:00 04F3:307A Touchpad"
P: Phys=i2c-ELAN1203:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-3/i2c-ELAN1203:00/0018:04F3:307A.0001/input/input25
U: Uniq=
H: Handlers=mouse1 event15 
B: PROP=1
B: EV=1b
B: KEY=e520 30000 0 0 0 0
--
I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="Asus Touchpad/Numpad"
P: Phys=
S: Sysfs=/devices/virtual/input/input46
U: Uniq=
H: Handlers=kbd event28 
B: PROP=0
`

tiberiufrat, 2022-11-03:
Same for me here, on a vivobook pro 14 oled
```$ grep -B1 -A5 -i -E "(elan|asu)" /proc/bus/input/devices
I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI hotkeys"
P: Phys=asus-nb-wmi/input0
S: Sysfs=/devices/platform/asus-nb-wmi/input/input9
U: Uniq=
H: Handlers=kbd event4 rfkill 
B: PROP=0
B: EV=100013
B: KEY=1000000080000 0 0 0 0 81606f00900000 c280027801701000 e000000000000 0
--
I: Bus=0018 Vendor=04f3 Product=31bc Version=0100
N: Name="ASUE140D:00 04F3:31BC Mouse"
P: Phys=i2c-ASUE140D:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE140D:00/0018:04F3:31BC.0001/input/input10
U: Uniq=
H: Handlers=event5 mouse0 
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=31bc Version=0100
N: Name="ASUE140D:00 04F3:31BC Touchpad"
P: Phys=i2c-ASUE140D:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE140D:00/0018:04F3:31BC.0001/input/input11
U: Uniq=
H: Handlers=event6 mouse1 
B: PROP=5
B: EV=1b
B: KEY=e520 10000 0 0 0 0
--
I: Bus=0018 Vendor=04f3 Product=31bc Version=0100
N: Name="ASUE140D:00 04F3:31BC Keyboard"
P: Phys=i2c-ASUE140D:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ASUE140D:00/0018:04F3:31BC.0001/input/input12
U: Uniq=
H: Handlers=sysrq kbd leds event9 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff800000000007ff febeffdfffefffff fffffffffffffffe
--
I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="Asus Touchpad/Numpad"
P: Phys=
S: Sysfs=/devices/virtual/input/input32
U: Uniq=
H: Handlers=kbd event24 
B: PROP=0```