mohamed-badaoui/asus-touchpad-numpad-driver #83: Pressing top left corner starts calculator, pressing top right corner enters equals sign

belegdol, 2021-12-11:
Hello,

on my UM425IA using m433ia layout, pressing top left corner starts the calculator, whereas pressing the top right corner enters the equals sign, but only when enabling the numpad.
Under Windows, one has to hold the top right corner to enable the numpad and swipe out the top left corner to start the calculator, which arguably reduces the chance of accidental activation. Would it be possible to implement similar behaviour for linux?

kawaegle, 2021-12-25:
hello , 
currently i'm on the testing because I find it's annoying sometime to enable the numpad.
for the moment I want to implement a timer for like 0.5 sec just to be sure that you really want to use the numpad and i'm going to add the slide down for the calculator so we can use the calculator in the numpad mode.

kawaegle

melons135, 2022-01-17:
Is there any way to top the '=' being entered when starting the numberpad?

melons135, 2022-01-17:
> Is there any way to top the '=' being entered when starting the numberpad?

i fixed this by chaning lines 107-110 to the following
`ai = d_t.absinfo[EV_ABS.ABS_X]
(minx, maxx) = (ai.minimum, ai.maximum - 200)
ai = d_t.absinfo[EV_ABS.ABS_Y]
(miny, maxy) = (ai.minimum + 100, ai.maximum)`

kawaegle, 2022-01-19:
yeah for the moment a solution by changing the activation away the "=" sign may work but 'll need to check how the windows version enable the numpad

btw thank you still open 'll closed when I check

857b, 2022-01-28:
Hello,
Another possible workaround for m433ia layout is to prevent the top margin from sending key events. asus-touchpad.py:271:
```py
col = math.floor(model_layout.cols * x / (maxx+1) )
row = math.floor((model_layout.rows * y / maxy) - model_layout.top_offset)
if row < 0: # add this test
    continue
try:
    button_pressed = model_layout.keys[row][col]
except IndexError: #...
```
Without this test, an input in the top margin is translated to `row = -1`, the last row of the layout, hence the '='.
Tested on UM425QA.

DanielProg39, 2022-02-05:
> Tested on UM425QA.

Hi, I have a question regarding overall functionality of asus numberpad in this model. Some guys have this issue: #75 . Do you personally face it?

dziedzic99, 2022-03-06:
Had this issue on ux371, fixed it and submitted a pull request. PR #90

belegdol, 2022-06-06:
This is fixed as of a2bada610ebb3fc002fceb53ddf93bc799241867