mohamed-badaoui/asus-touchpad-numpad-driver #85: is the calculator key annoying ?

kawaegle, 2021-12-25:
I want to know if you find the calculator key annoying and maybe if majority seem to say that it's annoying I'll try to set a little timer to have some delay to maintain the key before to launch the calculator

martyfrits, 2021-12-28:
A little bit of delay would be nice to have, but so far it didn't annoy me that much. The "=" sign typing on numpad activation is more annoying to me. If you could fix it somehow it would be nice, but otherwise you did an amazing job with this already. Thank you!

kawaegle, 2021-12-31:
you're welcome, but i'm only a little contributor

I'm trying to disable the calculator if you don't hold the top corner you just enable the numpad and not the calculator

and I don't know what enable the "=" sign but Iguess it's a problem with a double signal send to the event like when you press the top corner it send a signal that is also the "=" sign and so it's an "=" sign that appear

Kawaegle

axkibe, 2022-12-16:
For me it was annoying, I keep starting the calculator with my palm.

But given this is Python and seems to be well coded, it was easy to come up with a patch.

What I did was disabling the calculator start in normal touchpad mode, and make it start the calculator if in numpad mode.

Previously in numpad mode it would change brightness of the pad, I removed that and started always with max brightness.

Very happy with this, dunno if this reasonably can be integrated as start argument.

```--- asus_touchpad.py	2022-12-14 13:13:06.017951225 +0100
+++ /usr/share/asus_touchpad_numpad-driver/asus_touchpad.py	2022-12-16 17:39:21.872007152 +0100
@@ -200,7 +200,8 @@
 pos_x: int = 0
 pos_y: int = 0
 button_pressed: libevdev.const = None
-brightness: int = 0
+#brightness: int = 0
+brightness: int = 2
 
 while True:
     # If touchpad sends tap events, convert x/y position to numlock key and send it #
@@ -259,9 +260,13 @@
 
             # Check if caclulator was hit #
             elif (x < 0.06 * maxx) and (y < 0.07 * maxy):
+                #if numlock:
+                #    brightness = change_brightness(brightness)
+                #else:
+                #    launch_calculator()
+                #continue
                 if numlock:
-                    brightness = change_brightness(brightness)
-                else:
+                    #brightness = change_brightness(brightness)
                     launch_calculator()
                 continu
```

kawaegle, 2022-12-16:
Well IMO I want to totally disable the calculator launcher if we don't stay on the button for a certain time (I don't have access to the windows versions of this element but it was about 1 or 3 sec in my memory)

but actually thank you for giving a patch.

closed 2023-02-27 by kawaegle