mohamed-badaoui/asus-touchpad-numpad-driver #87: UX582LR Incorrect device detected

pckeller29, 2022-01-08:
Hi
First thanx for your amazing job, I learnt a lot with your code and it made my Zenbook even more pleasant to use!

I have a problem though, the device_id, touchpad and keyboard ids are incorrectly detected.

I'm using a Kubuntu 20.10 and the ux581l touchpad layout

Here is what I have when I check my input devices:

When I do a 
`egrep -B1 -A5 "ASUE|ELAN" /proc/bus/input/devices |grep -B1 -A5 Touchpad`

Here is what I get:
```
I: Bus=0018 Vendor=04f3 Product=2c23 Version=0100
N: Name="ELAN9009:00 04F3:2C23 Touchpad"
P: Phys=i2c-ELAN9009:00
S: Sysfs=/devices/pci0000:00/0000:00:15.3/i2c_designware.3/i2c-8/i2c-ELAN9009:00/0018:04F3:2C23.0003/input/input55
U: Uniq=
H: Handlers=mouse4 event258 
B: PROP=5
--
I: Bus=0018 Vendor=04f3 Product=3101 Version=0100
N: Name="ELAN1406:00 04F3:3101 Touchpad"
P: Phys=i2c-ELAN1406:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-ELAN1406:00/0018:04F3:3101.0004/input/input58
U: Uniq=
H: Handlers=mouse6 event17 
B: PROP=5
```
and of course, the first device_id that comes is 8 which is wrong for me. The correct one is 2.
Then, of course, the event id si wrong too: 258 is picked up while the correct one is 17
Same thing for the keyboard id, in my case, the correct one is 3.

```
I: Bus=0011 Vendor=0001 Product=0001 Version=ab83
N: Name="AT Translated Set 2 keyboard"
P: Phys=isa0060/serio0/input0
S: Sysfs=/devices/platform/i8042/serio0/input/input3
U: Uniq=
H: Handlers=sysrq kbd event3 leds 
B: PROP=0
```

So I did a dirty workaround for now (sorry) in your code, which works as far as I don't add another input device I guess:

```
# Start monitoring the touchpad

device_id='2'
keyboard=3
touchpad=17

...
```

Right now I try to figure out how to differentiate the correct device_id in a more global way. You probably have some good suggestion to help me improve this. For now, I'm not happy with my approaches.

Here is what I get in debug mode:

```
EBUG:Pad:Detect keyboard from N: Name="AT Translated Set 2 keyboard"
DEBUG:Pad:Set keyboard 3 from H: Handlers=sysrq kbd event3 leds
DEBUG:Pad:Detect touchpad from N: Name="ELAN9009:00 04F3:2C23 Touchpad"
DEBUG:Pad:Set touchpad device id 2 from S: Sysfs=/devices/pci0000:00/0000:00:15.3/i2c_designware.3/i2c-8/i2c-ELAN9009:00/0018:04F3:2C23.0003/input/input55
DEBUG:Pad:Set touchpad id 258 from H: Handlers=mouse4 event258
DEBUG:Pad:Touchpad min-max: x 0-1792, y 0-2500
```
Hope it will help, thx.

pckeller29, 2022-01-08:
Edit: After having re-read, I think the keyboard id is correctly detected. Will check this.

pckeller29, 2022-01-12:
I implemented a less dirty workaround by changing the search item:

 ```
          # Look for the touchpad #
            if touchpad_detected == 0 and "4F3:3101" in line and "Touchpad" in line:

```
Better than the previous one.

SiddhyN, 2022-01-23:
Same configuration for me (UX582LR) and your workaround work for me too @pckeller29 

Thx for the tips

(and thx to @mohamed-badaoui for the script !)

benderl, 2022-04-22:
Thanks! Also needed for UX582H.