mohamed-badaoui/asus-touchpad-numpad-driver #107: UX3402Z

r-schmitz11, 2022-12-22:
Hello,
I am contacting you because my keyboard is not available. I have the new asus zenbook 14 oled (ux3402z). I would like to know if it is possible to have the keyboard available too? 
I would like to know if it is possible to get it?
Romain

WhiteBlackGoose, 2023-01-12:
Hi, it works for me (NixOS). Couldn't get the backlight work yet, but keys on the touchpad work

ritchse, 2023-01-13:
it doesn't detect the touchpad at all for me

ritchse, 2023-01-13:
update: using the latest unstable debian got it to work in the same way as WhiteBlackGoose

t0m7, 2023-01-22:
Confirming it works like a charm for me, including backlight, on UX3402ZA running Ubuntu 22.04 LTS.

erad34, 2023-02-08:
It worked perfectly when I installed it a few days ago on Ubuntu 22.04 but the backlight has disappeared since. Very strange.

brueqlv, 2023-08-27:
Same, works, but without backlight.