mohamed-badaoui/asus-touchpad-numpad-driver #118: (Zenbook Flip UP5302Z) numpad works except backlight

Shogu, 2024-01-30:
Hi!

I tested a range of scripts and fixes, but i am now blocked.

Your driver is the best of all i tested (congratulations!), and it works fine on my Zenbook S Flip UP5302Z
Hi!

I tested a range of scripts and fixes, but i am now blocked.

Your driver is the best of all i tested, and it works fine on my Zenbook S Fliip UP5302Z

Unfortunately there is NO backlight, so the numpad i quite unusable :(

I tried the workarounds find here in the issues section (for example editing the *.py file with "change_brightness(1)"), but it fails.

Have you any idea?

Please note that i am not sure that this laptop is set up to have numpad, maybe the touchpad cannot be backlighted? The documentation is uncertain...
Unfortunately there is NO backlight, so the numpad i quite unusable :(

I tried the workarounds find here in the issues section (for example editing the *.py file with "change_brightness(1)"), but it fails.

Have you any idea?

Please note that i am not sure that this laptop is set up to have numpad, maybe the touchpad cannot be backlighted? The documentation is uncertain...

Thans in advance

Ogu from France

Shogu, 2024-02-19:
Hi again.

Is this project abandoned?

Thanks

Ogu

LittleBigOwI, 2024-04-18:
https://github.com/mohamed-badaoui/asus-touchpad-numpad-driver/issues/71

theT4mer, 2024-05-29:
Hi there, 
I had the same problem: numpad itself works like a charm but the backlight did not. I kinda solved it trought copy + pasting the "brightness = change_brightness(brightness)" line into the first if-condition:
```            # Check if numlock was hit #
            if (x > 0.95 * maxx) and (y < 0.09 * maxy):
                numlock = not numlock
                if numlock:
                    activate_numlock(brightness)
                    brightness = change_brightness(brightness)
                else:
                    deactivate_numlock()
                continue

            # Check if caclulator was hit #
            elif (x < 0.06 * maxx) and (y < 0.07 * maxy):
                if numlock:
                    brightness = change_brightness(brightness)
                else:
                    launch_calculator()
                continue
```
The this way, everything works perfectly and the light is triggerd when tapping the right icon. The left icon opens the system calculator. You can try if this works for you. I am however kind of hesitant to create a merge request because i don't understand the code fully and have done limited testing, so i might have caused some other weirdness with this change.
Greetings back from Germany!

dude2k5, 2025-01-25:
> Hi there, I had the same problem: numpad itself works like a charm but the backlight did not. I kinda solved it trought copy + pasting the "brightness = change_brightness(brightness)" line into the first if-condition:
> 
> ```
>             if (x > 0.95 * maxx) and (y < 0.09 * maxy):
>                 numlock = not numlock
>                 if numlock:
>                     activate_numlock(brightness)
>                     brightness = change_brightness(brightness)
>                 else:
>                     deactivate_numlock()
>                 continue
> 
>             # Check if caclulator was hit #
>             elif (x < 0.06 * maxx) and (y < 0.07 * maxy):
>                 if numlock:
>                     brightness = change_brightness(brightness)
>                 else:
>                     launch_calculator()
>                 continue
> ```
> 
> The this way, everything works perfectly and the light is triggerd when tapping the right icon. The left icon opens the system calculator. You can try if this works for you. I am however kind of hesitant to create a pull request because i don't understand the code fully and have done limited testing, so i might have caused some other weirdness with this change. Greetings back from Germany!

I can confirm this works! I thank you very much!